armbian/build #667: Partition too small. Need 3364 Mb Available:  Mb

Rombersoft, 2017-05-05:
I want to replace system from sd to sata on orange pi 2 plus.
It has 16 Gb on internal memory, 32 Gb on SD and 320 Gb on SATA.
I've run nand-sata-install command
selected http://joxi.ru/DrlNkGysvjZ0p2
next step http://joxi.ru/1A5y6MwIn71GBA
next step http://joxi.ru/Y2LyGbxI9DLEnA
next step http://joxi.ru/l2Z1DZjhwQN7zm
then console output:
mount: wrong fs type, bad option, bad superblock on /dev/sda1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
mount: wrong fs type, bad option, bad superblock on /dev/mmcblk1p1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
and in finish oops! http://joxi.ru/Q2KVMbXh4Ed4Zm

igorpecovnik, 2017-05-06:
Exactly which system do you use? Please provide `armbianmonitor -u`

Rombersoft, 2017-05-09:
root@orangepiplus:~# armbianmonitor -u
/var/log/armhwinfo.log has been uploaded to http://sprunge.us/gHXi
Please post the URL in the Armbian forum where you've been asked for.
root@orangepiplus:~#

zador-blood-stained, 2017-05-09:
`linux-jessie-root-orangepiplus` needs to be upgraded manually (with `apt install linux-jessie-root-orangepiplus`). System backup is highly recommended before the upgrade.

Rombersoft, 2017-05-10:
Ok I will try it later but before I want to copy data from SD to SATA. I've executed it via dd command but I suspect that there is wrong superblock on sda1 because I've tried mount sda1 and I got error about error with superblock.
Can you advise me a some right way to copy image from SD to SATA?

zador-blood-stained, 2017-05-10:
> Can you advise me a some right way to copy image from SD to SATA?

Using `nand-sata-install` is the correct way to move the rootfs. But you need to update it first since one of your screenshots showed some garbage instead of dialog options.

For the backup anything will work - dd, win32diskimager or other tools that can save full SD image just in case something breaks.

Rombersoft, 2017-05-10:
ok but does nand-sata-install command need formatted disk or with created partition?

igorpecovnik, 2017-05-10:
Recent version invoke fdisk if target is not initialized, otherwise you can choose the target format: ext2,3,4 and BTRFS if supported by kernel.

Rombersoft, 2017-05-11:
as for backup using dd command. I must create backup of whole drive or backup of partition? sda or sda1?

igorpecovnik, 2017-05-11:
sda1 or better: just backup critical files.

Rombersoft, 2017-05-11:
how do it with dd command (critical files)

igorpecovnik, 2017-05-11:
Not with DD but with tar,zip or something similar. For example: if you are running Mysql database server on the board you only backup databases (dump) and perhaps its mysql configuration. Nothing else.

Rombersoft, 2017-05-11:
I know that is mysql dump. But if my disk will break how will help me those
critical files? I think you want fool me.

igorpecovnik, 2017-05-11:
Disks do break and SD cards even more often. You always do regular backup to another media, to some USB key or remote location. What I am trying to tell you is that you don't need to backup whole disk drive.

Rombersoft, 2017-05-11:
but I need a bootable image, maybe I will want use it on another sd

igorpecovnik, 2017-05-12:
https://forum.armbian.com/index.php?/search/&q=sd%20card%20backup

closed 2017-05-12 by igorpecovnik

Rombersoft, 2017-05-12:
linux-jessie-root-orangepiplus has not resolved this problem:
http://joxi.ru/L21vYNwS8xM6lA
http://joxi.ru/GrqzNWqSQq5NXm
http://joxi.ru/Rmzeln6S0Ke0Yr
http://joxi.ru/ZrJNgbns9lB9Qr
http://joxi.ru/KAgk3Nnc4ad4eA
http://joxi.ru/LmG8NbRhe10eD2
http://joxi.ru/YmE87bZh0xy03r
mount: wrong fs type, bad option, bad superblock on /dev/sda1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
mount: wrong fs type, bad option, bad superblock on /dev/mmcblk1p1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.

http://joxi.ru/12MB0bkIMQzqpA

Thank for attention

igorpecovnik, 2017-05-12:
If there is a bug on our side, it's fixed for some time. I couldn't recreate this problem. http://sprunge.us/hTQO (eMMC boot / USB 4G key root)
I made an image right now, so try nightly images or upgrade to nightly and try again.

igorpecovnik, 2017-05-12:
Previous try was Ubuntu Xenial, I'll try with Jessie once again.

igorpecovnik, 2017-05-12:
Debian works too - update to nightly and it must work.

Rombersoft, 2017-05-13:
Hello Igor. There is one problem I think it may be related with this problem. So I can not mount manually sda1 partition of my SATA hdd with ext4 fs, error is "bad superblock..." and etc. But this SATA hdd is mounted now and working with fat32 fs. Do you know about this problem? As for nightly update I'll pass and wait a stability version. Thanks for attention. Have a nice day

igorpecovnik, 2017-05-13:
We certainly don't support installing to fat32 - this is a problem. Current installer should be able handle this ...

Rombersoft, 2017-05-13:
No no Igor. You did not understand me I simple can not execute mount /dev/sda1 /media with ext4 of sda1. But I can do it with fat32 of it

igorpecovnik, 2017-05-13:
Kill partition on your hard drive and use our tool to make it. Try with ext3. You need to start with stock image, upgraded to nightly and do not mount or format partitions by hand. This problem is not related to Armbian build system and so far I can't find relations to Armbian.

Click "new issue" and read the text to understand, why issue is closed.