armbian/build #669: banana pi m2 temp checking not working on 5.27

jkoenig72, 2017-05-06:
Hi,

love your work! Great material, great quality. 

Detected that with the latest build on the banana pi m2 the checking of temperature seems not to work anymore. 

cat /sys/devices/virtual/thermal/thermal_zone0/temp

Seems not to work any more, it did on 5.25.

igorpecovnik, 2017-05-06:
Known bug - will be fixed in next update.

zador-blood-stained, 2017-05-06:
> Seems not to work any more, it did on 5.25.

5.27 kernel was pushed to the stable repository by mistake. Please downgrade it manually.

@igorpecovnik 

I deleted the problematic kernel files from the repository without updating the index. IMO broken upgrades are better than working upgrades to broken versions.

jkoenig72, 2017-05-06:
Great ... I now this board is not the best and problematic but I have 3 here, so... :-)

Many thanks!

For now I exclude the kernel packages - seems fine to me.
dpkg --set-selections < kernalpackages.txt

Where kernalpackages.txt contains:

linux-headers-next-sunxi                        hold
linux-image-next-sunxi                          hold
linux-dtb-next-sunxi                            hold

igorpecovnik, 2017-05-07:
Index updated.

closed 2017-05-11 by zador-blood-stained