armbian/build #670: Realtek 8723bs module for Orange Pi Prime

amunozf, 2017-05-07:
According to the [official product site](http://www.orangepi.org/OrangePiPrime/), the Orange Pi Prime board has Realtek 8723bs chipset, but the module (rtl8723bs) is missing in the last nightly build (Armbian_5.27.170508_Orangepiprime_Ubuntu_xenial_dev_4.11.0.img)

The official Debian desktop image (debian_desktop_xfce_prime_V0_1.img), [downloaded from the Orange Pi website](http://www.orangepi.org/downloadresources/), has a working wifi chipset and I think It'll be nice to have a working wifi chipset in Armbian too.

Can you add this module in a future build?

zador-blood-stained, 2017-05-07:
Sources according to the wiki: https://github.com/jwrdegoede/rtl8723bs

Since it's an out of tree module it needs to be converted to a patch and tested, so it will take some time. Currently there are other issues for this board, so it will be at low priority unless somebody makes a pull request with required changes.

martinayotte, 2017-05-07:
Zador, I've just fixed the build for OPiPrime few hours ago, I'm now checking for other boards under 4.11.
As soon that I finish with those, I will take a look at this rl8723bs driver, at least to figure out how tedious it will be...

zador-blood-stained, 2017-05-07:
DVFS will be the main problem for H5 boards. There are a lot of clock related fixes and patches in the orange-pi-4.11 branch, + r_i2c patches (that may need migration to sunxi-ng to apply them on Icenowy's sunxi64-4.11.y). Compared to this 8723bs should be relatively easy, and we also need to add 8189es support too for some OPi boards.

martinayotte, 2017-05-07:
The 8189es diver is already working on OPiPC+ and OPiPlus2E with 4.11.
Which other boards needs some tweaks ?

zador-blood-stained, 2017-05-07:
8189**e**s is different from 8189**f**s
Orange Pi Lite, PC Plus and Plus 2E use 8189fs, 
Orange Pi 2, Plus and Plus 2 use 8189es and AFAIK it needs a separate driver

martinayotte, 2017-05-07:
Yes, I know they are not the same. But I've misread you post and I don't have those boards anyway ... :smile:

ap17, 2017-06-17:
Good day. 
zador-blood-stained commented on 7 May : "DVFS will be the main problem for H5 boards."
As it seems to me DVFS works good enough on 
IW's 4.10(.0)-rc8 (https://github.com/Icenowy/linux/tree/sunxi64-next-20170216) 
after applying your DVFS patches (build/patch/kernel/sun50i-dev/unresolved/), naturally, after some minor adaptations of these patches.
Is the problem (only) to transfer these patches and IW's made modifications of 4.10(.0)-rc8 to the currently used by you H5 kernel source 
(IW's 4.11.1, https://github.com/Icenowy/linux/tree/sunxi64-4.11.y) ?
Or are there any other "underwater stones"?
Excuse me for my misunderstandings of the problem.

ap17, 2017-06-19:
Good day. 
The following work has been done in regard of DVFS for H5.
1. The main patches [1] were applied to the currently used H5 kernel source [2].
2. A little modification of the DVFS patches [3] has been made and some new patch files were added to them.
3. Patches from step 2 were applied to the kernel source from step 1.
As a result DVFS, in my opinion, works well enough on OPi-Prime.
Tell me please how to present the modifications that has been made - may I present some of modified and new kernel files in the form of source files but not as patches ?
[1] : https://github.com/armbian/build/patch/kernel/sun50i-dev/
[2] : IW's 4.11.1, https://github.com/Icenowy/linux/tree/sunxi64-4.11.y
[3] : https://github.com/armbian/build/tree/master/patch/kernel/sun50i-dev/unresolved/

zador-blood-stained, 2017-06-19:
This is off-topic here, but there are following requirements to get H5 DVFS in Armbian:
- wait until Armbian migrates sunxi64 branch to kernel 4.12, where H5 support comes from upstream
- for the CPUfreq support, make use of clk notifier \[1\]
- for the THS and R_I2C  DT entries, use sunxi-ng bindings
- adapt sunxi-ng fixes from \[2\] if they are required

\[1\]: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/drivers/clk/sunxi-ng?id=02ae2bc6febd90cf3de61b6a1bdf491966ed410f
\[2\]: https://github.com/megous/linux/tree/orange-pi-4.11

ap17, 2017-06-19:
Thanks. Already has been done and works good on Opi-prime 4.11.6 .

zador-blood-stained, 2017-08-23:
H3/H5 dev branch was updated to 4.13. Rtl8723bs driver is now included by default, firmware will need to be added to firmware packages later.

closed 2017-08-23 by zador-blood-stained

Tonymac32, 2017-08-23:
RTL8723bs firmwares are in the packages, added for Tinker Board.

zador-blood-stained, 2017-08-23:
> RTL8723bs firmwares are in the packages, added for Tinker Board.

Yes, I just didn't find required files using `apt-file search` so they probably are not present in the stable repository yet.