armbian/build #673: [Nano PI Neo 2] Wrong DTS on u-boot & linux

hitech95, 2017-05-09:
I just build a custom Jessie image, but after looking the device tree from userpace looks like it loads the wrong one. The problem is also "present" on uboot.

The uboot config file specify an not existing DTS file. 
https://github.com/armbian/build/blob/master/patch/u-boot/u-boot-sun50i-dev/add-nanopineo2.patch#L13
The SBC loads as an Orange Pi PC 2, this problem looks like it propagate up to the Linux kernel:
```
INFO:    PSCI Affinity Map:
INFO:      AffInst: Level 0, MPID 0x0, State ON
INFO:      AffInst: Level 0, MPID 0x1, State ON
INFO:      AffInst: Level 0, MPID 0x2, State ON
INFO:      AffInst: Level 0, MPID 0x3, State ON

U-Boot SPL 2017.01-rc1-g5df570f-dirty (May 09 2017 - 10:42:28)
DRAM: 512 MiB
Failed to set core voltage! Can't set CPU frequency
Trying to boot from MMC1NOTICE:  BL3-1: Running on H5 (1718) in SRAM A2 (@0x44000)
NOTICE:  Configuring SPC Controller
NOTICE:  BL3-1: v1.0(debug):aa75c8d
NOTICE:  BL3-1: Built : 10:42:22, May  9 2017
INFO:    BL3-1: Initializing runtime services
INFO:    BL3-1: Preparing for EL3 exit to normal world
INFO:    BL3-1: Next image address: 0x4a000000, SPSR: 0x3c9

U-Boot 2017.01-rc1-g5df570f-dirty (May 09 2017 - 10:42:37 +0200) Allwinner Technology

CPU:   Allwinner H5 (SUN50I)
Model: OrangePi PC 2
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
*** Warning - bad CRC, using default environment

Setting up a 720x576i composite-pal console (overscan 32x20)
Error: no valid bmp image at 66000000
In:    serial
Out:   vga
Err:   vga
Net:   phy interface7
Could not get PHY for ethernet@1c30000: addr 1
No ethernet found.
Hit any key to stop autoboot:  0
38518 bytes read in 132 ms (284.2 KiB/s)
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
2428 bytes read in 208 ms (10.7 KiB/s)
## Executing script at 4fc00000
U-boot loaded from SD
Boot script loaded from mmc
116 bytes read in 169 ms (0 Bytes/s)
19253 bytes read in 325 ms (57.6 KiB/s)
4904371 bytes read in 531 ms (8.8 MiB/s)
11270152 bytes read in 743 ms (14.5 MiB/s)
## Loading init Ramdisk from Legacy Image at 4fe00000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    4904307 Bytes = 4.7 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
   reserving fdt memory region: addr=4fa00000 size=5000
   Loading Ramdisk to 49b52000, end 49fff573 ... OK
   Loading Device Tree to 0000000049b4a000, end 0000000049b51fff ... OK
Cannot setup simplefb: node not found

Starting kernel ...

Loading, please wait...
```

I tried to apply a my patch similar to this one:
https://patchwork.ozlabs.org/patch/760097/
but the SBC is still loade as OrangePi.

More info about my patch: https://gist.github.com/hitech95/61f88afa1aad633790e025f89d0d8a62
Regards, Nicolò.

hitech95, 2017-05-09:
@zador-blood-stained The problem is still present also after appling my patch, so i'm not sure what is the cause of the problem.

zador-blood-stained, 2017-05-09:
> i'm not sure what is the cause of the problem.

U-boot target in the Makefile and board config should be updated too. Main problem is that we are using a relatively old u-boot (2017.01) for 64-bit boards, so defconfig will need updates too.

hitech95, 2017-05-09:
> U-boot target in the Makefile and board config should be updated too.
Sadly I used to work with much older uboot versions. (Vendor specific, dated 2013 with of customs stuff)
Could you give me a start indication so i could fix? And maybe I'll try to create a complete patch.

zador-blood-stained, 2017-05-09:
The easiest way to handle this is leave everything as is, and DT file for the kernel can be set in u-boot environment/armbianEnv.txt

This Makefile defines the targets for the underlying u-boot and SPL: https://github.com/armbian/u-boot-sun50i/blob/master/Makefile

hitech95, 2017-05-09:
> The easiest way to handle this is leave everything as is, and DT file for the kernel can be set in u-boot environment/armbianEnv.txt

So you mean to use an overlay to handle the correct LED until u-boot is fixed/updated?
I did not see any parameters to directly specify the DTS path/address.

zador-blood-stained, 2017-05-09:
> I did not see any parameters to directly specify the DTS path/address.

`fdtfile` variable controls the DT file name for the kernel, and AFAIK kernel branch that we use (sunxi64-4.11.y) already has the DT for Neo 2
https://github.com/armbian/build/blob/master/config/bootscripts/boot-sun50i-next.cmd#L39

hitech95, 2017-05-09:
Oh, I did not really see that parameter, I've seen that there is already the DTS for the kernel.
Thanks for the information!

OT: just found a strange behaviour `shutdown -h now` reboot the board instead of freezing the SoC.

zador-blood-stained, 2017-08-23:
Neo 2 DT should be fixed by 2d20064258bc5741a424da6e9be1c8724914d7f9

closed 2017-08-23 by zador-blood-stained