armbian/build #676: Vagrant box doesnt build sdcard image

PTDreamer, 2017-05-11:
Mint 18 64bits host on 4.4.0-72 kernel with VirtualBox 5.1.22 r115126 (Qt5.5.1).

Shared folders are mounted with the nodev option so the build script fails on the debootstrap step complaining about the dir having the "noexec" or "nodev" options set.

pfeerick, 2017-05-30:
This _may_ be Mint or version specific - I don't have this issue with Ubuntu 16.04 on 4.4.0-78 kernel with VirtualBox 5.0.32_Ubuntu r11293 - I have generated two or three disk images now and they have run through to completions successfully. 

Not sure if it matters, but my lib folder is also in a subfolder of my home directory, so the ouput, build, user-patches, etc dirs are still created under my home directory, instead of one level up.

siedi, 2017-06-18:
Same happened to me (host is a Mac). The vagrant file creates a share for the output folder which is mounted with the nodev option. This causes the build process to fail. Removing the shared folder (so that the output is in the guest's filesystem) works fine.

commit 2e054a3c (2017-07-25): Improve Vagrant support
Fixes #676
commit 7f90f5d9 (2017-07-26): Improve Vagrant support
Fixes #676
commit 26252908 (2017-07-29): Improve Vagrant support
Fixes #676
commit 8396c953 (2017-08-01): Improve Vagrant support
Fixes #676
closed 2017-08-01 by zador-blood-stained