armbian/build #679: error localization

Rombersoft, 2017-05-12:
every time when I run command "apt-get upgrade" I get such warning:

perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:

Moreover I had noticed before that I can not type Cyrillic symbols in MySQL command prompt. After key pressing nothing happens. But in simple command prompt first typed Cyrillic symbol is not reflected correctly:

http://joxi.ru/4AkvxM1Synq512   (I've typed only letters but why I see <:> and <?>)

Also I see some bug with selecting text in terminal emulator. When I click left mouse button it achieves strange symbols:

http://joxi.ru/1A5y6MwIn7jJgA

kutysam, 2017-05-13:
Perhaps you can try the finetuning at here
https://docs.armbian.com/User-Guide_Fine-Tuning/

system language:

# Debian --> https://wiki.debian.org/ChangeLanguage
dpkg-reconfigure locales
# Ubuntu --> https://help.ubuntu.com/community/Locale
update-locale LANG=[options] && dpkg-reconfigure locales

Rombersoft, 2017-05-13:
Hello friend dpkg-reconfigure command is not found. I've tried to run apt-get install dpkg-reconfigure but it was not found there in repositories too. Maybe I need to add new repository?

igorpecovnik, 2017-05-13:
> every time when I run command "apt-get upgrade" I get such warning:
> dpkg-reconfigure command is not found.

On stock Armbian install this is working as expected - you must have broke something, you have failing SD card or you are using some non original Armbian build?

In any case -  let's move discussion to forum.armbian.com since this is not related to build process - where those issues tracker belong to. Since this is most likely general system issues, check sites like: https://askubuntu.com

Rombersoft, 2017-05-13:
OK. I test your project but problem reasons are in me. It is your congratulations. I want to donate your project but I've not gotten money for my projects which work on your Armbian and it is not stable yet.

martinayotte, 2017-05-13:
> Since this is most likely general system issues, check sites like: https://askubuntu.com
I agree ! Especially that every contributors receive emails about each post.
BTW, "-reconfigure" is an option to "dpkg", not a command. Simply put space before the dash.

zador-blood-stained, 2017-05-13:
> BTW, "-reconfigure" is an option to "dpkg", not a command. Simply put space before the dash.

```
➜  ~  % which dpkg-reconfigure
/usr/sbin/dpkg-reconfigure
➜  ~  %
```

martinayotte, 2017-05-13:
Sorry, my mistake ...

closed 2017-05-13 by igorpecovnik

Rombersoft, 2017-05-13:
OK, thank you

Rombersoft, 2017-05-13:
this has resolved problem in terminal emulator but has not resolved problem in mysql command prompt. symbols are not shown like I type password

igorpecovnik, 2017-05-13:
https://www.mysql.com/support

Rombersoft, 2017-05-13:
Ha ha
destructor@orangepiplus:~$ mysql --version
mysql  Ver 14.14 Distrib 5.6.30, for debian-linux-gnueabihf (armv7l) using  EditLine wrapper
Show me where is Oracle here? I have mysql on lubuntu and arch linux and linux mint without such problems

jobenvil, 2017-05-13:
Please @Rombersoft would you mind to move to the [armbian forum](https://forum.armbian.com/)? Like the NSA, many people out of the scene is watching you...