armbian/build #683: 4.11.0 DVFS / cpufreq-utils for orange pi zero

itdaniher, 2017-05-18:
Greetz!

Was delighted to see " Ubuntu server – mainline kernel" as an option for the OPZ on the most recent Armbian release. I grabbed a copy, but noticed cpufreq-info not returning anything useful. Based on `/sys/class/thermal/thermal_zone0/temp` I'm guessing it's running around 400MHz.

```
root@orangepizero:~# cpufreq-info
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.
```

Is this a known issue? 

FWIW `4.10.0-sun8i` nightly worked fine in this regard. I'm guessing a patch was left out / missed?

itdaniher, 2017-06-04:
```
root@orangepizero:~# cpufreq-info
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.
analyzing CPU 1:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.
analyzing CPU 2:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.
analyzing CPU 3:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.
root@orangepizero:~# uname -a
Linux orangepizero 4.11.3-sun8i #20 SMP Thu Jun 1 02:59:57 CEST 2017 armv7l armv7l armv7l GNU/Linux
```

This is still an issue with the latest nightly.

surfemodus, 2017-06-08:
Maybe this is the same problem I'm experiencing with OPi PC2 with ARMBIAN 5.27.170609 / 4.11.1-sun50iw2 updated today?

> $ cpufreq-info
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 1:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 2:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 3:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
$ uname -a
Linux opi 4.11.1-sun50iw2 #3 SMP Thu Jun 8 03:58:07 CEST 2017 aarch64 aarch64 aarch64 GNU/Linux

Noticed that rpi-monitor stopped showing cpu-frequency a while ago. Thought it was something I did..

itdaniher, 2017-06-08:
Similar kernel, similar CPU. Probably same problem.

Menion2k, 2017-06-28:
cpufreq is not supported in the Armbian used kernel
It exist a fork with proper support for it:
https://github.com/megous/linux
I am also pretty sure that, expecially with 4.11.6 the H5 is running very very slowly
Bye

surfemodus, 2017-06-28:
Where does it say that the H5 is running slowly in 4.11.6, as you say?

Menion2k, 2017-06-28:
I am as usual, making a tar gzip backup. before it tooks 2 hours, not it is running since 13 hours
And also sources build is very slow

zador-blood-stained, 2017-09-17:
OPi Zero DVFS should be fixed by 47e8f434bdf43525dbc998dcc58200d09b589722

closed 2017-09-17 by zador-blood-stained