armbian/build #684: Image built totaly different

blaster56, 2017-05-22:
I have build 2 images for my Orange Pi Pc + 

Build from source (server version), here the result : 

[http://imgur.com/a/RKSyR](url)

Name of 2 images i built : 

Armbian_5.27_Orangepiplus_Debian_jessie_default_3.4.113.img
Armbian_5.27_Orangepipcplus_Debian_jessie_default_3.4.113.img

I really dont understand what's happened...

igorpecovnik, 2017-05-23:
I don't understand your question - PI Plus and PI PC Plus are different boards.

closed 2017-05-28 by zador-blood-stained

blaster56, 2017-06-18:
Hello,

When i build from source with your script "non desktop version"and i flash the img from the output dir to my sd i have this on my tv screen : http://imgur.com/a/RKSyR.

I should have a CLI not a chineese strange GUI from Android its no sens !

ThomasKaiser, 2017-06-18:
Great idea, please provide a video of you burning the image you created with [Etcher](https://etcher.io) (that's the most important step -- you're not ready to accept that but together we will succeed). The video can be as short as 5 seconds, the only really important part is this screen here that must be there:

[image]

Unless your video does not contain this screen please repeat over and over again until you succeed!

martinayotte, 2017-06-18:
As said in forum thread, simply plug an USB-Serial and capture the boot log !
You will see that Armbian build isn't guilty of booting such Android image present in eMMC from manufacture.
Don't blame Armbian without knowing what you are doing ...