armbian/build #685: GCC 6 compile error when compile OrangePiLite Jessie

boddicheg, 2017-05-23:
Hey guys

Found interesting issue.

Type `./compile.sh`, then choose:
what to build: `OS image for installation to SD cart`
target: `orangepilite`
kernel branch: `default`
target os: `jessie`

and got error after some time 
```
 Generating include/generated/mach-types.h                                                                                                                                                               │  
  │   CC      kernel/bounds.s                                                                                                                                                                                 │  
  │ In file included from include/linux/compiler.h:48:0,                                                                                                                                                      │  
  │                  from include/linux/stddef.h:4,                                                                                                                                                           │  
  │                  from include/linux/posix_types.h:4,                                                                                                                                                      │  
  │                  from include/linux/types.h:17,                                                                                                                                                           │  
  │                  from include/linux/page-flags.h:8,                                                                                                                                                       │  
  │                  from kernel/bounds.c:9:                                                                                                                                                                  │  
  │ include/linux/compiler-gcc.h:106:30: fatal error: linux/compiler-gcc6.h: No such file or directory                                                                                                        │  
  │  #include gcc_header(__GNUC__)                                                                                                                                                                            │  
  │                               ^                                                                                                                                                                           │  
  │ compilation terminated.                                                                                                                                                                                   │  
  │ /home/boddicheg/workspace/Orange/sources/linux-sun8i/sun8i/./Kbuild:35: recipe for target 'kernel/bounds.s' failed                                                                                        │  
  │ make[1]: *** [kernel/bounds.s] Error 1                                                                                                                                                                    │  
  │ Makefile:988: recipe for target 'prepare0' failed                                                                                                                                                         │  
  │ make: *** [prepare0] Error 2                                                                                                                                                                              │  
  │ make: *** Waiting for unfinished jobs....          

[ error ] ERROR in function compile_kernel [ common.sh:238 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated 
                                                                     
```
Found [patch ](https://raw.githubusercontent.com/Freescale/meta-fsl-arm/master/recipes-kernel/linux/linux-ls1/Fix-the-compile-issue-under-gcc6.patch) that fix this problem(put in /lib/patch/kernel/sun8i-default )

zador-blood-stained, 2017-05-23:
Please provide file `output/debug/output.log`
`default` sun8i kernel is not supposed to be compiled with GCC6, so most likely you are using an unsupported build environment.

boddicheg, 2017-05-23:
[output.txt](https://github.com/armbian/build/files/1021570/output.txt)

zador-blood-stained, 2017-05-23:
```
Displaying message: Build host OS release stretch info
Displaying message: Host system support was not tested stretch wrn
Displaying message: Please don't ask for support if anything doesn't work
```
Debian Stretch is not supported as a build environment, sorry. Currently we are supporting only Ubuntu Xenial, as stated in the documentation and repository README, and fixing compilation of one kernel target is pointless without fixing and testing all other kernel and u-boot targets.

closed 2017-05-23 by zador-blood-stained