armbian/build #687: OPi+ 2E random MAC-address for WiFi causes trouble (dev-kernel)

WereCatf, 2017-05-30:
As the OPi+2E assigns a random MAC-address to the WiFi-device on every boot NetworkManager fails to bring up any associated WiFi-connections that were made with e.g. nmtui -- the connections are saved per MAC-address.

Perhaps generate a random MAC-address on the first boot, but then save the value somewhere and then assign it on subsequent boots to the device before NetworkManager is brought up?

zador-blood-stained, 2017-05-31:
It has to be fixed upstream (ideally) to use either MAC from DT via standard kernel APIs or to use SID directly as a hack. Don't know if anybody here is interested in fixing and maintaining this driver.

renekliment, 2017-10-01:
@WereCatf You can set a static MAC address to _spoof_ in NM so it stays the same. I have used this previously when I used Wi-Fi.

commit 54ac98c2 (2017-10-26): Read 8189fs MAC address from DT
Closes #687
closed 2017-10-26 by zador-blood-stained