armbian/build #688: Re-enable XR819 radio for orange pi zero

kutysam, 2017-05-31:
Hi, 
I understand that orange pi zero xr819 is not being compiled properly for 4.11.
Reason being, 4.11 adds a few more fields.
fifteenhex has enabled it for 4.11
https://github.com/fifteenhex/xradio/commit/3f80a7c69a4d00b93bb187edc6beef2b4c38bada

I hope this commit can be pushed. From what I see, it is not in, with reference to
https://github.com/armbian/build/blob/05fdf4192c612eff814f1c79824dd0a8045e638a/patch/kernel/sun8i-dev/add-xradio-wireless-driver.patch.disabled#L13601
Line 13601

I do understand that many have been yapping about the performance of the wireless chip.
In my opinion, let them yap. I am personally using it for wireless surviellence and it works perfectly! A video of 10fps is being transmitted smoothly over port 80.

Perhaps for those whom are yapping, you can put a note on your main website directing them to 
https://github.com/fifteenhex/xradio/blob/master/README.md

Thank you!

zador-blood-stained, 2017-05-31:
I'm against including XRadio driver by default. No matter how many warnings and where are added, people will still come and complain that something doesn't work.
If you want to use this driver on mainline you can still compile and install it as an out-of-tree module.

kutysam, 2017-05-31:
:( Is there any way to change your mind..?
I don't mind compiling and using it as an out-of-tree module.
However, I'm not too sure on how to do it. Could you give me some links or references.

After I've done it, I'll put it up for the community to use.

zador-blood-stained, 2017-05-31:
> Is there any way to change your mind..?

This driver either should work **perfectly** or it won't be included. In any case you won't be able to change the though process of people who have too high expectations for this hardware and who think that we are responsible for any issues. Since people think it's OK to talk like [here](https://forum.armbian.com/index.php?/topic/3997-orange-pi-zero-wifi-connection-issue/) I won't change my mind about this device.

> I don't mind compiling and using it as an out-of-tree module.
However, I'm not too sure on how to do it. Could you give me some links or references.

Generic documentation: https://docs.armbian.com/User-Guide_Advanced-Features/#how-to-build-a-wireless-driver

For this driver you may have to use something like `make M=$(pwd) KSRC=/usr/src/linux-headers-$(uname -r) modules` - or just check the README of the project https://github.com/fifteenhex/xradio/

kutysam, 2017-05-31:
Thank you for your reply. I'll work on the compilation soon.

Once again, I do hope that you will change your mind, or perhaps give an 'option' to enable wifi within armbian-config since this driver is 'experimental'

I do certainly appreciate armbian's effort in providing images for orange pi devices! 👍

closed 2017-05-31 by kutysam

rezaxdi, 2017-09-11:
This fork has an automatic installer and works pretty well for me : 
https://github.com/karabek/xradio

e1voronov, 2017-10-19:
Hi, I have loaded Armbian_5.34_Orangepizero_Ubuntu_xenial_next_4.13.8.7z (10-18-2017) to my OPZ.  
WiFi works very unstable with my TPLink router with the new image. 
The old armbian v5.25 (kernel 4.9.4) works more stable.

igorpecovnik, 2017-10-20:
> WiFi works very unstable with my TPLink router with the new image.

As you can see above, we are not very happy or satisfied either but will not even try to fix anything regarding this problem. If there will be more complaining, the driver will simply be disabled and support pleads ignored.

Armbian project is already low on resources, financed with our private resources and we simply can't afford to deal more on this https://financial-dictionary.thefreedictionary.com/Concorde+fallacy problem.

If you don't like the performance of this chip:

- throw the board away and buy the one with a good wireless chip,
- buy proven USB adaptor and continue to use it,
- fix the driver or raise money and (we will) establish "wireless driver mainlining team" to dig in more.

The old (3.4.113) kernel is officially EOL and will sooner or later completely lost support, so staying there is also not a good option.

e1voronov, 2017-10-20:
Dear Igor, that was just only my comment for the new 4.13.x. I know on the problems people are havinf with XR819.

itdaniher, 2017-10-20:
@igorpecovnik I've made several small anon donations to armbian in the past, and I'm interested in supporting your work. have you any idea what level of commitment would be required to support your "wireless driver mainlining team"? I'm interested in forking over some cash.

zador-blood-stained, 2017-10-20:
> have you any idea what level of commitment would be required to support your "wireless driver mainlining team"?

I don't think we have that kind of people in the team (and by "that kind" I mean people who can make or improve a driver from nothing when there is no documentation, no firmware fixes and no support from the vendor in general).

Icenowy has decided too that xradio is not worth the effort: https://irc.pine64.uk/?date=2017-10-20%2016:24:47

> I'm interested in forking over some cash.

It may be easier for you to spend that money on either a cheap USB wireless adapter or on a different board that has a good onboard wireless chip.

igorpecovnik, 2017-10-20:
> "wireless driver mainlining team"? I'm interested in forking over some cash.

I should put the whole idea under the quotes :) The only reward was frustration and that is not a good ground to start from. You can try with "pro"s: https://www.upwork.com (for example) Estimate 3 people, 100 hours x 50 EUR + 3rd party costs and there is zero warranty of success. You can go to companies which make drivers for business and they will give you some warranty but the price will jump. BTW. End-user donations for the whole project per year are of course (much) less than this.

mpmc, 2017-10-20:
@igorpecovnik 

> As you can see above, we are not very happy or satisfied either but will not even try to fix anything regarding this problem. **If there will be more complaining, the driver will simply be disabled and support pleads ignored.**

Even if you do disable the module and ignore support requests, I can guarantee people will continue to ask why the wifi on the OPi Z doesn't work. I personally hope you're thinking about removing the driver completely, not only because it's not great (not the developers fault) and causes more issues than it's worth, but use of wifi (on my board) increases the heat output to stupid levels.

Sadly **some** people don't read the "known issues" & download the images expecting everything to work. You could think about shifting the download links under the known issues, or maybe adding a popup before the download asking the user to confirm they've read the known issues section for their device.

igorpecovnik, 2017-10-20:
> I personally hope you're thinking about removing the driver 

This will sadly only change the title of the question. Why is the driver not there? :)

> Sadly some people don't read the "known issues" & download the images expecting everything to work. You could think about shifting the download links under the known issues, or maybe adding a popup [...]

I am not sure that will help. It's like in the traffic. When people get punished high enough and regularly than they perhaps change modus operandi. When something pops up that will search only for that checkbox - without reading. 

There will be some declutter on those download pages since they are not perfect, I agree, but I don't expect any major progress :(

Offtopic: Any CSS/HTML expert with spare time to help on WEB/WordPress themes redesign?

mpmc, 2017-10-20:
> This will sadly only change the title of the question. "Why is the driver not there?" :)

@igorpecovnik So basically a catch-22 :(

zador-blood-stained, 2017-10-20:
> You can try with "pro"s: https://www.upwork.com (for example) Estimate 3 people, 100 hours x 50 EUR + 3rd party costs and there is zero warranty of success. You can go to companies which make drivers for business and they will give you some warranty but the price will jump.

Or you can go to Allwinner and order XR819 chips from them (in quantities like 10k or more) and put some extra money on top of the deal - then maybe you could convince the business unit you are working this to make some software fixes, but even in this case you have a very low chance for success. I hope this will give an estimate on how much money is needed to get real fixes and not some placebo driver patching or randomly exchanging driver compilation options and hoping that anything will improve.

itdaniher, 2017-10-20:
Totally get it. No worries! The xr819 is definitely a mess, was asking as if you had a path forward but were limited by devtools or paid time. 

It's definitely a better path forward to included a well supported atheros part than the xr819. 

Thanks for your time and efforts!

gw0, 2017-11-03:
I am performing some WiFi testing on Orange Pi Zero with default installations of various distributions. All have a good USB power supply and are connected only to a WiFi AP 2.4 GHz WPA2. The results stay almost the same if SD cards are switched.

Description of columns:

- OS image name
- ping loss and average time at yesterday, 100x with 56B payload
- ping loss and average time at yesterday, 100x with 20kB payload
- ping loss and average time at today, 100x with 56B payload
- ping loss and average time at today, 100x with 20kB payload

```
Armbian_5.30_Orangepizero_Ubuntu_xenial_default_3.4.113    	0%/308	0%/459	1%/337	1%/522
Armbian_5.30_Orangepizero_Debian_jessie_default_3.4.113    	19%/590	100%/x	100%/x	100%/x
Armbian_5.34_Orangepizero_Ubuntu_xenial_next_4.13.8        	0%/76.4	1%/165	0%/75.2	0%/188
Armbian_5.34.171103_Orangepizero_Ubuntu_xenial_next_4.13.10	0%/75.6	0%/170	0%/69.8	0%/221
DietPi_OrangePiZero-armv7-(Jessie) (11-Mar-2017) + update  	24%/345	28%/593	28%/395	28%/575
Raspbian_server_For_zero_H2+_V0_1 (28-Apr-2017) + update   	0%/290	0%/435	0%/20.5	0%/152
```

As can be seen the next and nightly version of Armbian perform overall pretty good.