armbian/build #690: ad-hoc mode in 8919fs (rtl8919FTV) doesn't work

xvaara, 2017-05-31:
The ad-hoc mode doesn't work on orange pi lite or pcplus with the 8919fs driver.

I'm pretty sure it did work at one point last winter, but don't remember what kernel I was running.

```
root@5bff66:~# iwconfig
lo        no wireless extensions.

mesh-bridge  no wireless extensions.

wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

root@5bff66:~# iwconfig wlan0 mode ad-hoc essid test-mesh ap 02:12:34:56:78:9A channel 1
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Operation not permitted.
root@5bff66:~# ifconfig wlan0 up
root@5bff66:~# iwconfig wlan0 mode ad-hoc essid test-mesh ap 02:12:34:56:78:9A channel 1
root@5bff66:~# iwconfig
lo        no wireless extensions.

mesh-bridge  no wireless extensions.

wlan0     IEEE 802.11bg  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Ad-Hoc  Frequency:2.412 GHz  Cell: 02:11:87:00:AA:EA
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.
```

xvaara, 2017-05-31:
it seems that the 8919fs doesn't like when I try to set ad-hoc modes bssid to something. Is there someone somewhere that I can contact about this driver?

zador-blood-stained, 2017-05-31:
On mainline kernel we are using [this](https://github.com/jwrdegoede/rtl8189ES_linux) driver without any changes.
On legacy kernel - don't remember from where we got the driver.

In any case I don't think anyone will be fixing the legacy driver or the new driver until **and if** it is merged to mainline, especially since basic functionality (client/AP) should work AFAIK apart from random MAC address problem on mainline.

zador-blood-stained, 2017-05-31:
So what kernel version did you test - legacy or mainline? If ad-hoc mode works on one and doesn't work on the other it may be just a matter of compile-time configuration.

zador-blood-stained, 2017-05-31:
Mainline driver has `CONFIG_P2P_IPS` enabled and legacy driver has this option disabled. So please test both kernel versions.

xvaara, 2017-05-31:
I've traced the problem down to setting the bssid. It works if I run 
`iwconfig wlan0 mode ad-hoc essid test-mesh channel 1`
and it joins the previously started ad-hoc network with it's bssid. But it fails to set the bssid from iwconfig.

I'm running 4.11.1 and I did a quick test on legacy yesterday, but it didn't work either.

p2p and ad-hoc are different wlan modes.

closed 2017-10-25 by zador-blood-stained