armbian/build #691: Kernel linux-4.6.7 and Orange Pi PC patch

norrest, 2017-06-04:
Hi! 
Thank you for work! 
I need some another kernel for this board! 
I found based on 4.6.7 armbian 
Manually compile kernel, but after reboot lan dont work! ( 
May be need some patches for kernel? 
https://github.com/armbian/build/tree/master/patch/kernel
THANK YOU!

igorpecovnik, 2017-06-04:
H3 support is not fully done on mainline kernel and the current network driver is also getting a complete replacement while you should be able to build some semi workable image with our default DEV branch. We are using 4.11.x at this moment and network should work.

norrest, 2017-06-04:
so.. if I manually compile kernel  4.11.x  - after reboot all work fine?

norrest, 2017-06-04:
` CC      fs/xfs/xfs_dquot_item.o
  CC      fs/xfs/xfs_trans_dquot.o
  CC      fs/xfs/xfs_qm_syscalls.o
  CC      fs/xfs/xfs_qm_bhv.o
  CC      fs/xfs/xfs_qm.o
  CC      fs/xfs/xfs_quotaops.o
  CC      fs/xfs/xfs_rtalloc.o
  CC      fs/xfs/xfs_acl.o
  CC      fs/xfs/xfs_sysctl.o
  LD      fs/xfs/xfs.o
  LD      fs/xfs/built-in.o
  LD      fs/built-in.o
scripts/package/Makefile:91: recipe for target 'deb-pkg' failed
make[1]: *** [deb-pkg] Error 2
Makefile:1331: recipe for target 'deb-pkg' failed
make: *** [deb-pkg] Error 2
`

zador-blood-stained, 2017-06-04:
> so.. if I manually compile kernel 4.11.x - after reboot all work fine?

If you manually compile kernel 4.11 from [this](https://github.com/megous/linux/tree/orange-pi-4.11) branch it may work, but officially Ethernet will not be supported until release of stable kernel 4.13.

norrest, 2017-06-04:
**zador-blood-stained**
Thank you! 
Are this branch  support SPDIF?  
From armbian site:
_1-Wire protocol, reading out DHT11/DHT22 sensors or S/PDIF requires setting the minimum CPU frequency to 480MHz or higher_

zador-blood-stained, 2017-06-04:
> Are this branch support SPDIF?

It may work if properly activated in the Device Tree and if SPDIF pin is wired to the pin header.

norrest, 2017-06-04:
Like this ?
https://forum.armbian.com/index.php?/topic/1891-spdif-output-on-nanopi-m1/

norrest, 2017-06-04:
```
make oldconfig
make-kpkg clean
#make menuconfig
fakeroot make -j6 deb-pkg LOCALVERSION=-04062017.05 KDEB_PKGVERSION=1 
```
compile like this?

zador-blood-stained, 2017-06-04:
> Like this ?
https://forum.armbian.com/index.php?/topic/1891-spdif-output-on-nanopi-m1/

Yes, if using a freshly compiled or fresh nightly Armbian image.

> make oldconfig
make-kpkg clean
#make menuconfig
fakeroot make -j6 deb-pkg LOCALVERSION=-04062017.05 KDEB_PKGVERSION=1 
compile like this?

May work if you provide a good kernel config and patch the DT manually.
DT overlays will work only on fresh Armbian builds with Armbian (patched) kernel.

norrest, 2017-06-04:
patch the DT manually ? 
What patch are need ? 
Also i used patch MuQSS, but 
```

                 from ./include/asm-generic/bug.h:13,
                 from ./arch/arm/include/asm/bug.h:59,
                 from ./include/linux/bug.h:4,
                 from ./include/linux/thread_info.h:11,
                 from ./include/asm-generic/current.h:4,
                 from ./arch/arm/include/generated/asm/current.h:1,
                 from ./include/linux/sched.h:11,
                 from kernel/sched/MuQSS.c:33:
./arch/arm/include/asm/ptrace.h:31:10: error: invalid type argument of ‘->’ (hav                                                                                        e ‘int’)
  (((regs)->ARM_cpsr & 0xf) == 0)
          ^
kernel/sched/MuQSS.c:3097:6: note: in expansion of macro ‘user_mode’
  if (user_mode(get_irq_regs()))
      ^
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target 'kernel/sched/MuQSS.o' failed
make[4]: *** [kernel/sched/MuQSS.o] Error 1
scripts/Makefile.build:553: recipe for target 'kernel/sched' failed
make[3]: *** [kernel/sched] Error 2
Makefile:1002: recipe for target 'kernel' failed
make[2]: *** [kernel] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC      drivers/ata/libata-scsi.o
  CC      drivers/ata/libata-eh.o
  CC [M]  crypto/vmac.o
  CC      fs/btrfs/file-item.o
  CC [M]  crypto/xcbc.o
  CC [M]  crypto/rmd128.o
  CC      fs/btrfs/inode-item.o
  CC [M]  fs/ceph/snap.o
  CC      drivers/ata/libata-transport.o
  CC      mm/vmpressure.o
  LD      fs/cifs/built-in.o
  CC [M]  crypto/rmd160.o
  CC [M]  fs/cifs/cifsfs.o
  CC      fs/btrfs/inode-map.o
  CC [M]  fs/ceph/xattr.o
  CC      drivers/ata/libata-trace.o
  CC      mm/swap_cgroup.o
  CC [M]  crypto/rmd256.o
  CC      mm/cleancache.o
  CC      fs/btrfs/disk-io.o
  CC      drivers/ata/libata-sff.o
  CC [M]  fs/cifs/cifssmb.o
  CC [M]  fs/ceph/mds_client.o
  CC      mm/page_isolation.o
  CC [M]  crypto/rmd320.o
  CC      mm/zpool.o
  CC      drivers/ata/libata-pmp.o
  CC      mm/zbud.o
  CC [M]  crypto/sha512_generic.o
  CC      mm/zsmalloc.o
  CC      drivers/ata/ahci_platform.o
  CC [M]  crypto/wp512.o
  CC [M]  fs/ceph/mdsmap.o
  CC      fs/btrfs/transaction.o
  CC      drivers/ata/libahci.o
  CC      mm/early_ioremap.o
  CC [M]  fs/cifs/cifs_debug.o
  CC [M]  crypto/tgr192.o
  CC [M]  fs/ceph/strings.o
  CC      mm/cma.o
  CC [M]  fs/ceph/ceph_frag.o
  CC [M]  crypto/pcbc.o
  CC [M]  fs/ceph/debugfs.o
  CC      mm/frame_vector.o
  CC [M]  fs/cifs/connect.o
  CC      fs/btrfs/inode.o
  CC      drivers/ata/libahci_platform.o
  LD      mm/built-in.o
  CC [M]  crypto/lrw.o
  CC      drivers/ata/ahci_sunxi.o
  LD [M]  fs/ceph/ceph.o
  CC      fs/btrfs/file.o
  CC      fs/configfs/inode.o
  LD      drivers/ata/libata.o
  LD      drivers/ata/built-in.o
  LD      drivers/atm/built-in.o
  LD      drivers/auxdisplay/built-in.o
  CC      drivers/base/component.o
  CC [M]  crypto/keywrap.o
  CC      fs/configfs/file.o
  CC      drivers/base/core.o
  CC [M]  crypto/gcm.o
  CC [M]  fs/cifs/dir.o
  CC      fs/configfs/dir.o
  CC [M]  crypto/ccm.o
  CC      fs/configfs/symlink.o
  CC      fs/configfs/mount.o
  CC [M]  fs/cifs/file.o
  CC      fs/configfs/item.o
  CC      drivers/base/bus.o
  CC [M]  fs/cifs/inode.o
  LD      fs/configfs/configfs.o
  LD      fs/configfs/built-in.o
  CC      fs/debugfs/inode.o
  CC [M]  crypto/chacha20poly1305.o
  CC      drivers/base/dd.o
  CC      fs/debugfs/file.o
  CC [M]  crypto/pcrypt.o
  CC      drivers/base/syscore.o
  CC      fs/btrfs/tree-defrag.o
  CC [M]  fs/cifs/link.o
  LD      fs/debugfs/debugfs.o
  CC      drivers/base/driver.o
  LD      fs/debugfs/built-in.o
  CC      drivers/base/class.o
  CC [M]  crypto/cryptd.o
  CC      fs/btrfs/extent_map.o
  CC      fs/btrfs/sysfs.o
  LD      drivers/bcma/built-in.o
  CC [M]  drivers/bcma/main.o
  CC      drivers/base/platform.o
  CC [M]  crypto/mcryptd.o
  CC [M]  fs/cifs/misc.o
  CC [M]  fs/cifs/netmisc.o
  CC [M]  drivers/bcma/scan.o
  CC      fs/btrfs/struct-funcs.o
  CC      drivers/base/cpu.o
  CC [M]  crypto/fcrypt.o
  CC [M]  drivers/bcma/core.o
  CC      drivers/base/firmware.o
  CC [M]  crypto/blowfish_generic.o
  CC [M]  fs/cifs/smbencrypt.o
  CC [M]  fs/cifs/transport.o
  CC      fs/btrfs/xattr.o
  CC      drivers/base/init.o
  CC [M]  drivers/bcma/sprom.o
  CC [M]  crypto/blowfish_common.o
  CC      drivers/base/map.o
  CC      drivers/base/devres.o

```

norrest, 2017-06-04:
wget http://ck.kolivas.org/patches/muqss/4.0/4.11/4.11-sched-MuQSS_156.patch 
patch -p1 < 4.11-sched-MuQSS_156.patch

zador-blood-stained, 2017-06-04:
> patch the DT manually ?
What patch are need ?

If not using prebuilt Armbian images with Armbian kernel, you'll need to find examples how to activate SPDIF in the Device Tree for your board and patch it either before the kernel compilation ([example](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/arch/arm/boot/dts/sun8i-h3-beelink-x2.dts?id=ef09e0f539f124c67aa44d4154730bc8492196bf)) or afterwards.

> Also i used patch MuQSS

We can't support 3rd party patches and backports (especially on development/experimental kernel branch), so it's up to you to resolve any problems and conflicts.

norrest, 2017-06-04:
I am using 
**Welcome to ARMBIAN Debian GNU/Linux 8 (jessie) 4.6.7-sun8i**

zador-blood-stained, 2017-06-04:
> 4.6.7-sun8i

Which is too old. SPDIF is supported starting with 4.11, which is the current version for Armbian development branch.

norrest, 2017-06-04:
Ok!
```
mount /dev/sda1 /mnt
sudo git clone https://github.com/megous/linux.git /mnt/linux-megos
cd /mnt/linux-megos
make oldconfig
make-kpkg clean
#make menuconfig
fakeroot make -j6 deb-pkg LOCALVERSION=-04062017.05 KDEB_PKGVERSION=1

```

Are this enjoyed to install new kernel or spdif ?

zador-blood-stained, 2017-06-04:
> Are this enjoyed to install new kernel and spdif ?

No.
I already said several times what is needed - either precompiled Armbian image where DT overlays can be used, or compiling orange-pi-4.11 branch with proper kernel config and patching the DT. And we can'n provide good support or step by step instructions for any of those options untill we have stable Armbian releases with this kernel.

norrest, 2017-06-04:
Why can not this kernel be installed on top of ARMBIAN Debian GNU / Linux 8 (jessie) 4.6.7-sun8i? (

zador-blood-stained, 2017-06-04:
I checked Orange Pi PC schematics and you won't be able to use onboard SPDIF out on this board without soldering because pin PA17 (the only one that can be used) is wired to the CSI (camera) header. Sorry.

You may be able to use USB sound cards with SPDIF output, but it's up to you to find a sound card with good support in the kernel that you are using.

closed 2017-06-04 by zador-blood-stained

norrest, 2017-06-04:
> s wired to the CSI (camera) header 
How to unwire?

norrest, 2017-06-04:
I can soldering) No problems!

zador-blood-stained, 2017-06-04:
Schematics: https://drive.google.com/file/d/0BwQT0FP7IkTbRFJQamNwWTAzWms/view
Pin PA17 is wired to pin 4 of CSI connector (flat connector between button and microUSB port) as CSI-PWR-EN

norrest, 2017-06-08:
**zador-blood-stained** 
Thank YOU!
Is there no way to redirect PA17 to another port? 
In order not to solder