armbian/build #693: Nano PI Neo2 + broken ipsec

jjssoftware, 2017-06-05:
This is a known issue in 4.11 which I believe the Neo2 mainline experimental build for the AllWinner H5 is based on.

ipsec is reported broken in 4.11 and the fix is being tracked here: https://patchwork.kernel.org/patch/9704017/

I'm raising this issue here just as a question about process really; given where this patch is up to at the moment, what's the expected / estimated timeline for a patch like this to eventually find it's way into Armbian mainline?

Many thanks
Joe

zador-blood-stained, 2017-06-05:
This patch is in linux-next ([link](https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/net/ipv4/esp4.c?id=0e78a87306a6f55b1c7bbafad1de62c3975953ca)) and linux-stable master ([link](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/net/ipv4/esp4.c?id=0e78a87306a6f55b1c7bbafad1de62c3975953ca)), but it didn't reach linux-4.11.y yet.

We should add this patch to **stable** branches if it doesn't reach linux-4.11.y by the time we are doing a stable release.

jjssoftware, 2017-06-05:
Thanks Mikhail that sounds great for Armbian stable.

I can tell from the [changelog](https://docs.armbian.com/Release_Changelog/) that Armbian is on 4.11 but it doesn't say specifically what 4.11.x level Armbian mainline is current at. Whilst trying to debug this issue I've moved off Armbian for moment so I'm not sure what 4.11.x level Armbian mainline is current at.

I do know this issue is present in kernel 4.11.2 and given 4.11.3 is released and this specific fix is not yet released this bug must also be present in 4.11.3.

Is there any word what happens to us if we're on an experimental build :) Does Armbian release a more detailed changelog anywhere that I could possibly watch to be notified when this fix makes it into the experimental mainline?

zador-blood-stained, 2017-06-05:
> Is there any word what happens to us if we're on an experimental build :) Does Armbian release a more detailed changelog anywhere that I could possibly watch to be notified when this fix makes it into the experimental mainline?

With the nightlies it's more complicated - even though they are rebuilt every day, we are using custom branches for H3, H5 and A64 based boards, and those branches are not updated frequently. There is no detailed changelog for the kernels, so for this fix you'll have to monitor version diffs at kernel.org and then compare Armbian kernel version to the upstream version that got the fix.

Or you can just compile a kernel manually with this patch included and test by yourself if it fixes the problem.

jjssoftware, 2017-06-05:
> Or you can just compile a kernel manually with this patch included and test by yourself if it fixes the problem

If the process is as easy as it sounds as outlined in the docs then I think this is my way forward. So long as the build script enables build of a mainline experimental for the H5 all should be good..

It feels like I'm going to be having a fun weekend 😊

jjssoftware, 2017-06-10:
I believe this is where the sunxi kernel source gets pulled from by the Armbian build system:

https://github.com/Icenowy/linux/tree/sunxi64-4.11.y

If so, this is the file where the patch will be applied:

https://github.com/Icenowy/linux/blob/sunxi64-4.11.y/net/ipv4/esp4.c

I'd like to get this patch in there.

Do anyone know who maintains https://github.com/Icenowy/linux/tree/sunxi64-4.11.y ?
Is this maintained by linux core kernel developers?

[kernel-sun50iw2-dev.patch.zip](https://github.com/armbian/build/files/1065767/kernel-sun50iw2-dev.patch.zip)

jjssoftware, 2017-06-11:
Thanks for https://github.com/armbian/build/commit/1080d5f8400c9b65dec1fbff20cdeaa1052c4a8f

I'll admit I don't understand the positioning of this patch because I don't understand enough about the Armbian build system but I trust you've done the right thing.

ThomasKaiser, 2017-06-12:
> I'll admit I don't understand the positioning of this patch

You ignore the difference between **stable** and experimental (**no** support).

jjssoftware, 2017-06-17:
No I don't ignore anything, that's a wrong statement / assumption. I also don't expect any special support or special treatment when using the Neo2 H5 experimental build. I do however note the recent change to armbian.com that makes the status of no support more apparent so well done there..
```diff
- Work in progress: experimental images, no end user support
```
I suppose I should stop talking here now to prevent this also being incorrectly interpreted as a support request.

zador-blood-stained, 2017-08-23:
H5/A64 dev branch was switched to 4.13 release candidates. Currently there are other issues (including possible stability problems), but this IPsec problem should be solved.

closed 2017-08-23 by zador-blood-stained