armbian/build #696: RAID6_PQ should be disabled by default

PTDreamer, 2017-06-12:
RAID6_PQ causes a boot delay of 3.5s on nanopineo. Furthermore even with menuconfig it is not obvious what should be disabled  because since there is no help text associated with this configuration there is no easy was of knowing what other configurations depend on it. 
`Location:                                                                                                                                                     │  
  │ (2) -> Library routines                                                                                                                                         │  
  │   Defined at lib/Kconfig:10                                                                                                                                     │  
  │   Selected by: MD_RAID456 [=m] && MD [=y] && BLK_DEV_MD [=m] || ORE [=n] && (EXOFS_FS [=n] || PNFS_OBJLAYOUT [=n]) || BTRFS_FS [=y] && BLOCK [=y]             `

zador-blood-stained, 2017-06-12:
"Selected by" shows exactly why this symbol was enabled.
And it's a hidden symbol, so it can't be controlled manually - it will be enabled automatically if any other option depends on it.

zador-blood-stained, 2017-06-12:
Also: #205

PTDreamer, 2017-06-12:
> "Selected by" shows exactly why this symbol was enabled.
> And it's a hidden symbol, so it can't be controlled manually - it will be enabled automatically if any other option depends on it.

The selected by doesn't show up because there is no help text on the kconfig. The text posted above was after adding some bogus help text.

zador-blood-stained, 2017-06-12:
> The selected by doesn't show up because there is no help text on the kconfig. The text posted above was after adding some bogus help text.

An easy way to find hidden options is using the menuconfig "search" function (`/` key).
Example with an unmodified kernel:
```
Symbol: RAID6_PQ [=y]
Type  : tristate
  Defined at lib/Kconfig:10
  Selected by: MD_RAID456 [=m] && MD [=y] && BLK_DEV_MD [=m] || ORE [=n] && (EXOFS_FS [=n] || PNFS_OBJLAYOUT [=n]) || BTRFS_FS [=y] && BLOCK [=y]

PTDreamer, 2017-06-12:
> An easy way to find hidden options is using the menuconfig "search" function (/ key).

Full Ack

closed 2017-06-12 by PTDreamer