armbian/build #697: Nano PI Neo AIR new model w/ new WiFi revision

splitice, 2017-06-16:
Newly ordered NanoPI hardware comes with a new revision of the Wifi Chip "A". It's now a AP6212**A** (following lines are: W3112005 1706), On boot WiFi no longer works, dmesg reports a timeout communicating with the chip.

Logs from the device will follow shortly.

andrew-payne, 2017-06-16:
```
root@NewNanoPiAir:/home/npa dmesg | grep -i wifi
[    1.455948] [rf_pm]: set losc_out 32k out[wifi_pm]: set wl_reg_on 1 !
[    1.656590] [wifi_pm]: wifi gpio init is OK !!
[    9.184635] dhd_wifi_platform_load: Enter
[    9.184718] wifi_platform_set_power = 1
[    9.184751] [wifi_pm]: set wl_reg_on 0 !
[    9.384822] [wifi_pm]: set wl_reg_on 1 !
[    9.890199] wifi_platform_bus_enumerate device present 1
[   22.210116] dhd_wifi_platform_load_sdio: sdio_register_driver timeout or error 
[   22.210338] wifi_platform_set_power = 0
[   22.210378] wifi_platform_bus_enumerate device present 0
[   22.210495] unregister wifi platform drivers
[   22.210517] wifi_platform_bus_enumerate device present 0
[   22.210757] dhd_wifi_platform_load: Enter
[   22.210832] wifi_platform_set_power = 1
[   22.210866] [wifi_pm]: set wl_reg_on 0 !
[   22.410942] [wifi_pm]: set wl_reg_on 1 !
[   22.920099] wifi_platform_bus_enumerate device present 1
[   35.260110] dhd_wifi_platform_load_sdio: sdio_register_driver timeout or error 
[   35.260336] wifi_platform_set_power = 0
[   35.260376] wifi_platform_bus_enumerate device present 0
[   35.260493] unregister wifi platform drivers
[   35.260514] wifi_platform_bus_enumerate device present 0
[   35.260761] dhd_wifi_platform_load: Enter
[   35.260838] wifi_platform_set_power = 1
[   35.260876] [wifi_pm]: set wl_reg_on 0 !
[   35.460952] [wifi_pm]: set wl_reg_on 1 !
[   35.970105] wifi_platform_bus_enumerate device present 1
[   48.260112] dhd_wifi_platform_load_sdio: sdio_register_driver timeout or error 
[   48.260330] wifi_platform_set_power = 0
...
```
Above log is from Armbian 5.25 Debian Jessie.
We just discovered the new 5.30 release, will try that out.

splitice, 2017-06-16:
Fixed on the latest armbian :) 👍

closed 2017-06-16 by splitice