armbian/build #699: lircd stopped working (most probably after upgrade to Linux 4.11 kernel)

muwlgr, 2017-06-19:
Earlier, lircd startup looked like this : 
lircd-0.9.4c[7151]: Notice: accepted new client on /var/run/lirc/lircd
lircd-0.9.4c[7151]: Info: initializing 'auto'
lircd-0.9.4c[7151]: Info: Using device: /dev/input/event1

After reboot with updated Linux kernel :
Linux version 4.11.5-sunxi (root@devel) (gcc version 7.1.1 20170510 (Linaro GCC 7.1-2017.05) ) #3 SMP Thu Jun 15 02:11:01 CEST 2017
it looks like : 
lircd-0.9.4c[1115]: Notice: lircd(devinput) ready, using /var/run/lirc/lircd
lircd[1634]: lircd: there seems to already be a lircd process with pid 1115
lircd-0.9.4c[1115]: Notice: accepted new client on /var/run/lirc/lircd
lircd-0.9.4c[1115]: Info: initializing '/dev/input/'
lircd-0.9.4c[1115]: Info: Using device: /dev/input/
lircd-0.9.4c[1115]: Warning: can't get exclusive access to events coming from `/dev/input/' interface
lircd-0.9.4c[1115]: Error: error reading '/dev/input/'
lircd-0.9.4c[1115]: Info: closing '/dev/input/'

and after that, systemd spawns new lircd processes which spew the same records in the log : error reading '/dev/input/' , closing '/dev/input/'

in /etc/lirc/hardware.conf, we see :
str=$(cat /proc/bus/input/devices | grep "H: Handlers=sysrq rfkill kbd event" | awk '{print $(NF)}')
sed -i 's/DEVICE="\/dev\/input.*/DEVICE="\/dev\/input\/'$str'"/g' /etc/lirc/hardware.conf

but with kernel 4.11, in /proc/bus/input/devices we only see 
H: Handlers=kbd event0 
H: Handlers=kbd event1 
H: Handlers=kbd event2 
without any 'sysrq' and 'rfkill' words
so we get empty string 'str', and the only DEVICE="/dev/input/" in this config which is of course directory name, not any special file.

any thought on how to fix this ?
I don't use lirc features, so I can safely remove lirc package.
but what about others who do ?

zador-blood-stained, 2017-06-19:
> any thought on how to fix this ?
I don't use lirc features, so I can safely remove lirc package.

This will be fixed by using udev rules instead of executing commands in hardware.conf to find the correct input device. Unfortunately it's not possible to create a good udev rule for the legacy 3.4.x kernel, but on mainline it can be done easily.

zador-blood-stained, 2017-10-08:
Problematic hardware.conf detection logic was removed from the build scripts, so this can be closed.

closed 2017-10-08 by zador-blood-stained