armbian/build #700: Audio is broken on H3 (OrangePi PC) after upgrade to Linux 4.11 mainline kernel

jojo141185, 2017-06-19:
My OrangePi PC is on ARMBIAN 5.31 with mainline kernel 4.11.5-sun8i

jochen@orangepi:~$ dmesg | grep codec
_[   13.112542] rc rc0: lirc_dev: driver ir-lirc-codec (sunxi-ir) registered at minor = 0
[   13.169891] sun4i-codec 1c22c00.codec: ASoC: /soc/codec-analog@01f015c0 not registered
[   13.180498] sun4i-codec 1c22c00.codec: Failed to register our card
[   13.456851] sun4i-codec 1c22c00.codec: ASoC: /soc/codec-analog@01f015c0 not registered
[   13.467500] sun4i-codec 1c22c00.codec: Failed to register our card
[   13.793389] sun4i-codec 1c22c00.codec: Codec <-> 1c22c00.codec mapping ok_

jochen@orangepi:~$ aplay -l
_**** List of PLAYBACK Hardware Devices ****
card 0: Codec [H3 Audio Codec], device 0: CDC PCM Codec-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0_

Tried to fix that by adding a user overlay with newest conent from repo [https://github.com/armbian/sunxi-DT-overlays/blob/master/sun8i-h3/sun8i-h3-analog-codec.dts]:
jochen@orangepi:~$ sudo armbian-add-overlay sun8i-h3-analog-codec.dts
jochen@orangepi:~$ sudo nano /boot/armbianEnv.txt
_overlay_prefix=sun8i-h3
overlays=spdif-out cir
user_overlays=sun8i-h3-analog-codec_
jochen@orangepi:~$ sudo sync
jochen@orangepi:~$ sudo reboot

Unfortunately nothing changed and error still persists :(
Any thoughts on how to fix this ? I need working hdmi audio with a newer kernel version >4.10.

igorpecovnik, 2017-06-20:
Mainline support is at experimental level and AFAIK there are no HDMI audio drivers - you need to create them or wait. Only essentials works. Take a peek here: http://linux-sunxi.org/Linux_mainlining_effort

zador-blood-stained, 2017-06-20:
HDMI audio is not supported on mainline

closed 2017-06-20 by zador-blood-stained

jojo141185, 2017-06-20:
Thanks, for the quick answer igorpecovnik and zador-blood-stained. 
I had a look at the matrix on [this page](http://linux-sunxi.org/Linux_mainlining_effort) befor I upgraded from legacy to mainline. There I see Audio Codec (H3) is supported in mainline Kernel >4.10 - probably as you mentioned without HDMI support. But under WIP section I found this [repo](https://github.com/jernejsk/linux-1/commit/5d109ee4f802a46839c6d3cdc99121c0fd5991a3) of Jernej Skrabec with a DT-overlay to enable HDMI controller on Orange-Pi PC. 

Even if HDMI output will not work on mainline kernel, why do I get en error on loading the analog-codec DT-overlay? 

Thank you for any help in advance.

zador-blood-stained, 2017-06-20:
"Audio codec" means analog audio - 3.5mm jack, microphone and possibly speakers on some boards/configurations. I2S audio interface, for example, is handled by another driver (not implemented on H3), SPDIF output is handled by yet another driver (should work), and HDMI audio also pre-depends on DRM video driver and HDMI support in the status matrix.

DRM driver for H3 is not done yet and it's not present in the branch that is used for the H3 dev kernel.

Errors in the log are related to deferred probe of analog audio output support, and as `aplay -l` shows, analog audio driver is loaded correctly.

jojo141185, 2017-06-20:
I see. Thanks zador-blood-stained, for your detailed explanation. 
So I will switch back to legacy kernel and hope these features will be supported in neer future. Keep my fingers crossed! I am grateful to the armbian team for all their excellent work.