armbian/build #705: u-boot patches not all applied

Menion2k, 2017-06-28:
Hi all
I am working on having a working cpufreq support for Allwinner H5 SoC
According to the mainline plan, it won't happen anytime soon (it is not even scheduled for 4.13)
However some (working) patches already exist, there is a quite updated fork at:
https://github.com/megous/linux
Reading the provided information, there is also an absolutely required patch for u-boot that fix a PLL1 configuration
Apparently this patch is already included in Armbian, it is the patch:

> lib/patch/u-boot/u-boot-sun50i-dev/pll1-clock-fix-h5.patch

But for some reason I really don't understand, this is not applied to the u-boot source, the file 

> sources/u-boot-sun50i/master/u-boot/arch/arm/mach-sunxi/clock_sun6i.c

doesn't contain the patch. Other patches in u-boot-sun50i-dev are applied correctly, so I am wondering why (and how) this particular one is out.
I don't have any userpatches override
Bye

Menion2k, 2017-06-28:
Hi
I have just noticed that the DVFS and THS patches for H5 have been moved to the "unresolved" directory
Can it be due to kernel lookup caused by the missing u-boot patch above?

zador-blood-stained, 2017-08-03:
I just tested this and `pll1-clock-fix-h5.patch` is applied correctly.

zador-blood-stained, 2017-08-03:
As for DVFS and THS - it was already discussed on the forum.

closed 2017-08-03 by zador-blood-stained