armbian/build #711: Wifi Driver issue in armbian-firmware 5.30

ryanm101, 2017-07-08:
clean install of Orange Pi Zero Plus 2 H5: Armbian_5.27_Orangepizeroplus2-h5_Ubuntu_xenial_dev_4.11.0
```
$> apt-get -s install armbian-firmware
The following packages will be upgraded:
  armbian-firmware
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Inst armbian-firmware [5.27] (5.30 Armbian:xenial [arm64])
Conf armbian-firmware (5.30 Armbian:xenial [arm64])
```

As soon as the system reboots wlan0 disappears from ifconfig
checking with lshw show no network entry.
Prior to upgrade lshw shows:
```
 *-network
       description: Wireless interface
       physical id: 9
       logical name: wlan0
       serial: XX:XX:XX:XX:XX:XX
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=brcmfmac driverversion=0.7.46.57 firmware=es6.c5.n4.a3 ip=X.X.X.X multicast=yes wireless=IEEE 802.11
```

lsmod (prior to upgrade output)
```
root@orangepizeroplus2:~# lsmod
Module                  Size  Used by
ir_xmp_decoder         16384  0
ir_mce_kbd_decoder     16384  0
ir_sharp_decoder       16384  0
ir_sanyo_decoder       16384  0
ir_sony_decoder        16384  0
ir_jvc_decoder         16384  0
ir_rc6_decoder         16384  0
ir_nec_decoder         16384  0
ir_rc5_decoder         16384  0
bluetooth             356352  0
sun4i_codec            49152  3
sun8i_codec_analog     28672  1
ir_lirc_codec          16384  0
snd_soc_core          147456  2 sun4i_codec,sun8i_codec_analog
lirc_dev               20480  1 ir_lirc_codec
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               102400  2 snd_pcm_dmaengine,snd_soc_core
snd_timer              32768  1 snd_pcm
sunxi_cir              16384  0
rc_core                32768  13 ir_sanyo_decoder,ir_rc6_decoder,ir_nec_decoder,ir_mce_kbd_decoder,ir_xmp_decoder,ir_sony_decoder,ir_lirc_codec,ir_sharp_decoder,lirc_dev,ir_jvc_decoder,sunxi_cir,ir_rc5_decoder
brcmfmac              212992  0
brcmutil               16384  1 brcmfmac
cfg80211              266240  1 brcmfmac
rfkill                 28672  5 bluetooth,cfg80211
```

lsmod after
```
Module                  Size  Used by
ir_xmp_decoder         16384  0
ir_mce_kbd_decoder     16384  0
ir_sharp_decoder       16384  0
ir_sanyo_decoder       16384  0
ir_sony_decoder        16384  0
ir_jvc_decoder         16384  0
ir_rc6_decoder         16384  0
ir_nec_decoder         16384  0
ir_rc5_decoder         16384  0
bluetooth             360448  0
ir_lirc_codec          16384  0
lirc_dev               20480  1 ir_lirc_codec
sun8i_codec_analog     28672  1
sun4i_codec            49152  3
snd_soc_core          147456  2 sun4i_codec,sun8i_codec_analog
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               102400  2 snd_pcm_dmaengine,snd_soc_core
sunxi_cir              16384  0
snd_timer              32768  1 snd_pcm
rc_core                32768  13 ir_sanyo_decoder,ir_rc6_decoder,ir_nec_decoder,ir_mce_kbd_decoder,ir_xmp_decoder,ir_sony_decoder,ir_lirc_codec,ir_sharp_decoder,lirc_dev,ir_jvc_decoder,sunxi_cir,ir_rc5_decoder
brcmfmac              217088  0
brcmutil               16384  1 brcmfmac
cfg80211              274432  1 brcmfmac
rfkill                 28672  3 bluetooth,cfg80211
```

tokio69, 2017-07-19:
I also have the same issue. I use a OrangePi Zero Plus2 H5

messi89, 2017-07-24:
I have the same issue after apt upgrade on my  OPi Zero plus2 H5

messi89, 2017-07-24:
any fix for this bug?

or we need to rollback the upgrade ?

igorpecovnik, 2017-07-25:
There will be no urgent fix since this image is work in progress / development, while reporting bug gives us important information that our approach for handling different wireless chips with the same driver is not working perfectly fine and we need to try some other way. Perhaps somebody change something in kernel to break our fix. Can't tell before investing time into this issue. For possible workaround scenarios, put AP6212A into https://forum.armbian.com/index.php?/search/

messi89, 2017-09-02:
is there any update about this issue ?

Nexxkinn, 2017-09-03:
I'd suggest you to just buy a cheap wifi dongle and us that one instead.

zador-blood-stained, 2017-10-24:
Should be fixed by https://github.com/armbian/firmware/commit/3aa265d38307115a8be673aeaf16ec36439b6d06

closed 2017-10-24 by zador-blood-stained

messi89, 2017-10-26:
so we can update our Orange Pi Zero Plus 2 H5 now ? any one try it ?

igorpecovnik, 2017-10-26:
Try it. You need to use beta.armbian.com repository.

messi89, 2017-10-26:
@igorpecovnik  on the apt source.list ?

igorpecovnik, 2017-10-26:
In /etc/apt/sources.list.d/armbian.list change apt to beta and conduct update & upgrade. If you fail ... start with a clean image, especially if you are running some older build. All current images are beta, so an upgrade is generally untested. https://dl.armbian.com/orangepizeroplus2-h5/Ubuntu_xenial_next_nightly.7z Wifi should work, while BT is AFAIK not enabled yet.

Nexxkinn, 2017-10-28:
@igorpecovnik Any bugs should I expect about the nightly version on OPI Zero other than BT ?

igorpecovnik, 2017-10-28:
> All current images are beta, so an upgrade is generally untested.

Thi

ryanm101, 2017-11-08:
@messi89  Did it work?

Nexxkinn, 2017-11-12:
@ryanm101 yep, it works.

messi89, 2017-11-18:
after upgrade pi zero won't boot...I will download a new version :(

messi89, 2017-11-18:
whats happen with the last nighty build https://dl.armbian.com/orangepizeroplus2-h5/

bootloop... I will install the old version and try to update

igorpecovnik, 2017-11-19:
> whats happen with the last nighty build https://dl.armbian.com/orangepizeroplus2-h5/

It works, tested right now. It must be some problem on your side. Bad SD card, PSU or cable.

messi89, 2017-11-19:
I don't have problem with sd card (original kingstone) and I have installed the os on the emmc

igorpecovnik, 2017-11-19:
I didn't try to install onto eMMC but will later. Please open a topic on the forum and provide all possible data (logs, screenshot, ...) since it's hard to help/find problems. I am locking this thread since it is completely offtopic.