armbian/build #712: Disable dialog for compilation and debootstrap operations?

zador-blood-stained, 2017-07-09:
Any thoughts/objections against setting `PROGRESS_DISPLAY=plain` by default or removing `PROGRESS_DISPLAY=dialog` at all?

- while testing compilation in Docker environment I noticed that dialog has some rendering glitches with Docker terminal
- dialog window may hide warning messages in some situations
- it's harder to find and copy compilation error messages when using the dialog front-end
- apt-get in current debootstrap implementation displays progress and a graphical progress bar, which is not possible to use with dialog

igorpecovnik, 2017-07-09:
I would vote for changing defaults.

ThomasKaiser, 2017-07-09:
I second that, let's make `PROGRESS_DISPLAY=plain` the default.

zador-blood-stained, 2017-07-09:
OK, I'll push the changes later, possibly with other improvements and cleanups for the `compile.sh`

closed 2017-07-09 by zador-blood-stained

martinayotte, 2017-07-09:
Yes ! As a matter of fact, I'm using PROGRESS_DISPLAY=plain since almost a year ... ;-)

zador-blood-stained, 2017-07-09:
> Yes ! As a matter of fact, I'm using PROGRESS_DISPLAY=plain since almost a year ... ;-)

I used it since... since I implemented it :smile: 
IMO dialog based input (build options) is good, but dialog based output (logs) - not so much