armbian/build #713: Use guestfish to create disk images (get rid of filesystem mount requirement)

jadr2ddude, 2017-07-13:
There is a tool called [guestfs](http://libguestfs.org/guestfish.1.html) that is meant to create VM images without mounting on the host. It supports generating raw disk images. This would be nice for armbian because then it will be possible to get rid of the requirement of root access/docker privileged mode.

zador-blood-stained, 2017-07-13:
> it will be possible to get rid of the requirement of root access/docker privileged mode.

And how do you propose to handle chroot and systemd-nspawn without superuser privileges?

jadr2ddude, 2017-07-13:
Maybe fakechroot if that works. Otherwise getting rid of docker priveliged mode is good enough.

zador-blood-stained, 2017-07-13:
> Maybe fakechroot if that works.

For compiling even fakeroot works. For debootstrap - I highly doubt.

> Otherwise getting rid of docker priveliged mode is good enough.

It may be possible to use non-privileged mode and pass loop devices and maybe extra capabilities to `docker run`. I'll try to test this as a part of ongoing refactoring process.

zador-blood-stained, 2017-07-13:
https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities

jadr2ddude, 2017-07-13:
I am currently trying to see if I can get proot to work with debootstrap. I think I have figured out a way to get it to work correctly. I am going to try it tomorrow.

jadr2ddude, 2017-07-13:
Or: https://wiki.debian.org/EmDebian/CrossDebootstrap#Generating_cross_images_as_non-root_user

zador-blood-stained, 2017-07-14:
Uh, building images in non-privileged Docker is more complicated than I thought, but seems that using Docker Edge may have a solution to some of the problems, I'll test it next.

zador-blood-stained, 2017-08-02:
Now it should be possible to build full OS images in Docker (pretty recent Docker verion is required, ones in Debian/Ubuntu repositories are pretty old).

So since you don't need root privileges to run Docker, this is kind of solved.

closed 2018-04-29 by zador-blood-stained

probonopd, 2018-10-24:
I am trying to build in an unprivileged Docker instance provided by GitLab CI. Despite having specified `NO_APT_CACHER=yes` the build fails when trying to create the images with

```
Processing triggers for systemd (237-3ubuntu10.3) ...
Processing triggers for initramfs-tools (0.130ubuntu3.5) ...
update-initramfs: Generating /boot/initrd.img-4.18.16-odroidc1
cryptsetup: WARNING: failed to detect canonical device of /dev/mmcblk0p1
cryptsetup: WARNING: could not determine root device from /etc/fstab
cryptsetup: WARNING: failed to detect canonical device of /dev/mmcblk0p2
Warning: couldn't identify filesystem type for fsck hook, ignoring.
update-initramfs: Converting to u-boot format
linux-u-boot-odroidc1-next set on hold.
linux-image-next-odroidc1 set on hold.
[ o.k. ] Preparing image file for rootfs [ odroidc1 bionic ]
[ o.k. ] Current rootfs size [ 733 MiB ]
[ o.k. ] Creating blank image for rootfs [ 888 MiB ]
[ o.k. ] Creating partitions [ root: ext4 ]
[ error ] ERROR in function check_loop_device [ image-helpers.sh:73 ]
[ error ] Device node /dev/loop0 does not exist 
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:60 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
ERROR: Job failed: exit code 1
```

Am I missing something? Would using something like http://libguestfs.org/guestmount.1.html instead of loop be an option?