armbian/build #719: Crypto acceleration on A20 via sun4i-ss

Tanarri, 2017-07-19:
Hi,

I am using a cubietruck with A20 cpu, and my HTTPS performance is bad. (HTTP is fast)

Would it be possible to add crypto acceleration (e.g. AES) with the sun4i-ss kernel modul?
Here I read about it:
http://sunxi.montjoie.ovh/
https://linux-sunxi.org/Security_System 

Greetings
Eugen

jadr2ddude, 2017-07-19:
We tried to add this for the H3, which works the same way. The issue is that you need OpenSSL 1.1.0 for this to work, which is too new for many Ubuntu/Debian packages.

jadr2ddude, 2017-07-19:
However, you can probbably prioritize CHACHA20_POLY1305 (depending on application) which is quite fast.

zador-blood-stained, 2017-07-19:
sun4i-ss kernel module is already present and working in the kernel for A20 devices. Based on benchmarks it's not that fast anyway - it's more offloading than acceleration. And whether software uses it or not is out of our control.

closed 2017-08-01 by zador-blood-stained