armbian/build #720: finux-firmware-image-sun* firmware packages conflict with firmware-linux-free (on stretch)

whohoho, 2017-07-24:
The following NEW packages will be installed:
  linux-firmware-image-sun4i linux-firmware-image-sun7i linux-firmware-image-sun8i
0 upgraded, 3 newly installed, 0 to remove and 9 not upgraded.
Need to get 508 kB of archives.
After this operation, 2171 kB of additional disk space will be used.
Get:1 http://apt.armbian.com jessie/main armhf linux-firmware-image-sun4i armhf 5.31 [112 kB]
Get:2 http://apt.armbian.com jessie/main armhf linux-firmware-image-sun7i armhf 5.31 [284 kB]
Get:3 http://apt.armbian.com jessie/main armhf linux-firmware-image-sun8i armhf 5.31 [112 kB]
Fetched 508 kB in 0s (992 kB/s)                 
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
/usr/bin/locale: Cannot set LC_CTYPE to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_MESSAGES to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_ALL to default locale: No such file or directory
E: Can not write log (Is /dev/pts mounted?) - posix_openpt (19: No such device)
Selecting previously unselected package linux-firmware-image-sun4i.
(Reading database ... 18283 files and directories currently installed.)
Preparing to unpack .../linux-firmware-image-sun4i_5.31_armhf.deb ...
Unpacking linux-firmware-image-sun4i (5.31) ...
dpkg: error processing archive /var/cache/apt/archives/linux-firmware-image-sun4i_5.31_armhf.deb (--unpack):
 trying to overwrite '/lib/firmware/keyspan_pda/xircom_pgs.fw', which is also in package firmware-linux-free 3.4
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Selecting previously unselected package linux-firmware-image-sun7i.
Preparing to unpack .../linux-firmware-image-sun7i_5.31_armhf.deb ...
Unpacking linux-firmware-image-sun7i (5.31) ...
dpkg: error processing archive /var/cache/apt/archives/linux-firmware-image-sun7i_5.31_armhf.deb (--unpack):
 trying to overwrite '/lib/firmware/keyspan_pda/xircom_pgs.fw', which is also in package firmware-linux-free 3.4
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Selecting previously unselected package linux-firmware-image-sun8i.
Preparing to unpack .../linux-firmware-image-sun8i_5.31_armhf.deb ...
Unpacking linux-firmware-image-sun8i (5.31) ...
dpkg: error processing archive /var/cache/apt/archives/linux-firmware-image-sun8i_5.31_armhf.deb (--unpack):
 trying to overwrite '/lib/firmware/keyspan_pda/xircom_pgs.fw', which is also in package firmware-linux-free 3.4
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/linux-firmware-image-sun4i_5.31_armhf.deb
 /var/cache/apt/archives/linux-firmware-image-sun7i_5.31_armhf.deb
 /var/cache/apt/archives/linux-firmware-image-sun8i_5.31_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

# dpkg -s firmware-linux-free
Package: firmware-linux-free
Status: install ok installed
Priority: optional
Section: kernel
Installed-Size: 68
Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>
Architecture: all
Multi-Arch: foreign
Source: firmware-free
Version: 3.4

zador-blood-stained, 2017-08-03:
These packages are a byproduct of the kernel build process and are not meant to be installed. And trying to install multiple kernel firmware packages won't work anyway since they will conflict with each other.

`armbian-firmware` and `armbian-firmware-full` are the supported packages, and since there is no official support for Stretch their dependencies were not checked yet, it will be done later.

zador-blood-stained, 2017-10-19:
@igorpecovnik 
Since in-kernel firmware will be dropped starting with mainline 4.14 (and since it was pretty old and irrelevant long before that) it makes sense to remove all linux-firmware-* packages from packaging scripts and our repositories, this way we wouldn't have to worry about dependencies for anything except armbian-firmware and upstream Debian/Ubuntu firmware packages.

igorpecovnik, 2017-10-20:
So you mean we only pack armbian-firmware without -full in here: https://github.com/armbian/build/blob/master/packages/extras/firmware.sh

zador-blood-stained, 2017-10-20:
I meant leaving armbian-firmware as is but remove all linux-firmware-image-* packages (even older ones) from the repository and update the repository index.
These packages for example: [[1]](http://apt.armbian.com/pool/main/l/linux-4.11.4-cubox/linux-firmware-image-next-cubox_5.30_armhf.deb), [[2]](http://apt.armbian.com/pool/main/l/linux-4.4.68-mvebu/linux-firmware-image-mvebu_5.31_armhf.deb), [[3]](http://apt.armbian.com/pool/main/l/linux-4.9.0-sunxi/linux-firmware-image-dev-sunxi_5.22_armhf.deb)

igorpecovnik, 2017-10-20:
aha. ok. Will clean up. Do we need this per arch? https://github.com/armbian/build/blob/master/packages/extras/firmware.sh#L35 any

zador-blood-stained, 2017-10-20:
> Do we need this per arch? 

Not really, it can be set to "all"

igorpecovnik, 2017-10-20:
Removed all excessive garbage, firmware, all DEV pkgs, prehistoric builds, ... 5.4G -> 1.7G I hope I didn't delete too much :) Added XU4 latest build ...

zador-blood-stained, 2017-10-28:
Can be closed by https://github.com/armbian/build/commit/97d01dbe162bdccbe61f744e6e0bf8c2d4b7799f

closed 2017-10-28 by zador-blood-stained