armbian/build #721: Documentation lacks information on EXPERT flag

ecolyx, 2017-07-24:
After building several images (for various Orange boards), I obtained an experimental board, but could not find information on building an experimental image using the EXPERT="yes" flag, without searching the forums.

If a link to docs can be provided I will seek to contribute the solution (add a section on EXPERT mode to the docs).

igorpecovnik, 2017-07-25:
This information is hidden by purpose. We don't want to expose expert targets to random user since they are flooding us with support questions from areas, which are not supported: work in progress, deprecated boards and community supported boards.

ecolyx, 2017-07-25:
Thanks for response. I guess I found what I needed, sorry for adding to noise.

closed 2017-07-26 by ThomasKaiser