armbian/build #725: Cleaning up the Vagrantfile.

gabegundy, 2017-08-03:
There are many nice changes to the Vagrantfile, but I wonder if the provision script is _too_ much. Currently, it installs packages that will need to installed and links a few directories for output and userpatches.

I don't think it's a good idea to install all those packages when bringing up the vm for the first time. Don't those packages get installed as part of the build process anyway? Installing via the provisioning scripts means there are now two places where you have to track the packages that are required.

I purpose we limit the Vagrantfile to things that might be Vagrant-specific and let the build script take care of the rest.

I've happy to do this, but didn't want to step on anyone's toes.

gabegundy, 2017-08-03:
I'll put my other nit-picks on here 😉 

Removing `vb.gui = true` makes the process much more opaque to users that are new to Vagrant and Armbian. It's already a lot to wrap one's head around the first time you build.

zador-blood-stained, 2017-08-03:
> I don't think it's a good idea to install all those packages when bringing up the vm for the first time. Don't those packages get installed as part of the build process anyway? Installing via the [...]

This type of provisioning makes sense for Docker which caches intermediate containers. Removing packages installation from Vagrant provisioning makes sense for me. I didn't experiment much with Vagrant (yet).

> Removing vb.gui = true makes the process much more opaque to users that are new to Vagrant and Armbian. It's already a lot to wrap one's head around the first time you build.

I saw this situation as users who are new to Armbian building will use plain VirtualBox, and Vagrant will be used for people who are already familiar with it (especially since Vagrant by itself has no GUI and no hardware compatibility checks so I can't say that it is newbie friendly). I'm fine with reverting to `vb.gui = true` by default.

gabegundy, 2017-08-03:
Do we prefer to clone the code when we have a copy of the repo in the vm's /vagrant? We could sync it to /home/ubuntu/armbian.

BTW, the symlinks don't seem to be working for me, but I'll keep playing with it.

Sorry, not trying to be a pain, just tying to make sure I understand the direction we're going so I can help most.

zador-blood-stained, 2017-08-03:
> We could sync it to /home/ubuntu/armbian.

And lose file permissions, symlinks and possibly enforce other limitations of Virtualbox shared directories.

> Sorry, not trying to be a pain, just tying to make sure I understand the direction we're going so I can help most.

The direction is - update this stuff to work with the new directory structure, ensure that it is transparent to users and that it is well documented (I'm planning to make separate documentation pages for each supported build method with clearly stated hardware and software requirements).

BTW, now it's possible to build full OS images in non-privileged Docker containers, which means that on Debian, Ubuntu and other OS supported by Docker it may be an easier and more lightweight building method with lesser system requirements.

Also Vagrant is compatible with Windows, which adds some filesystem related limitations - no symlinks or file permissions in the git project directory, and even `git archive` won't create symlinks if they are not enabled for git (which may require additional privileges).

JonahBraun, 2017-09-03:
Hi @gabegundy,

> Removing vb.gui = true makes the process much more opaque to users that are new to Vagrant and Armbian. It's already a lot to wrap one's head around the first time you build.

I appreciate what you are trying to do here. However the entire point of using Vagrant is to allow developers NOT to have to concern themselves with seeing up a virtual machine. The boot console is not something people need to look at, it doesn't provide any useful information. `vagrant up` already provides status information.

For those familiar vagrant based project, having the console pop up is slightly annoying. I'm not sure how much you've worked with VirtualBox, but working with the VirtualBox console is not something you want to be doing. It's interaction is restricted and awkward compared to ssh in your favorite terminal.

More importantly however this breaks builds on headless servers meaning that those of us who are running automated test servers now need to maintain a fork just to work around this.

There's definitely room to make things user friendly, this switch isn't one.  :)

lanefu, 2019-03-06:
closing old issue

closed 2019-03-06 by lanefu