armbian/build #726: debootstrap-ng fails at stage 2/2 of installing base system

WereCatf, 2017-08-04:
debootstrap-ng fails at stage 2/2 of installing base system at line 151 when building a development-kernel, desktop Xenial-image for OPi PC unless PROGRESS_LOG_TO_FILE is set to "yes." Didn't try with other boards or distros.

Build host is Ubuntu Xenial 16.04 x64 in a virtual machine.

zador-blood-stained, 2017-08-04:
Logs? Console output?

zador-blood-stained, 2017-08-04:
Just tested - stage 2 works for me.

WereCatf, 2017-08-04:
I just tested and it fails the exact same way as it did last night: https://pastebin.com/1RDvD5kr

Like I mentioned, enabling logs fixes the fail.

zador-blood-stained, 2017-08-04:
These logs don't have any meaningful messages. Please provide **console output** starting with the second debootstrap stage.

WereCatf, 2017-08-04:
I didn't include it because it doesn't provide any more useful information, but whatever: https://pastebin.com/YzjUfzdn

zador-blood-stained, 2017-08-04:
That's double strange. Looks like debootstrap returns a non-zero exit code for no apparent reason, and also ${PIPESTATUS[0]} depends on other commands in the pipe.
And at least I can't reproduce this and there are no other similar bug reports.

Did you try removing everything and starting from scratch?

WereCatf, 2017-08-04:
Obviously, yes. The OS is a clean install, with nothing extra other than what the Armbian build-system installs itself, too.

igorpecovnik, 2017-08-04:
I am also rebuilding everything and haven't noticed a thing. Huh.

WereCatf, 2017-08-04:
I have talent for hitting all sorts of weird corner-cases 😨

zador-blood-stained, 2017-08-04:
Well, you may be able o debug this by changing the code around line 151 to something like this
```
display_alert "Installing base system" "Stage 2/2" "info"
eval 'chroot $SDCARD /bin/bash -c "/debootstrap/debootstrap --second-stage"' \
	${PROGRESS_LOG_TO_FILE:+' | tee -a $DEST/debug/debootstrap.log'} \
	${OUTPUT_DIALOG:+' | dialog --backtitle "$backtitle" --progressbox "Debootstrap (stage 2/2)..." $TTY_Y $TTY_X'} \
	${OUTPUT_VERYSILENT:+' >/dev/null 2>/dev/null'}

[[ ${PIPESTATUS[0]} -ne 0 || ! -f $SDCARD/bin/bash ]] && echo "Debootstrap base system second stage failed"

echo "Debootstrap paused, press Enter to continue"
echo "Debootstrap directory path is ${SDCARD}/debootstrap/"
read

mount_chroot "$SDCARD"
```
there should be debootstrap logs in the /debootstrap directory, maybe they have some useful info

WereCatf, 2017-08-05:
I figured it out, the issue came from the underlying file-storage method -- I don't know why, exactly, but nevertheless, the issue got solved.

closed 2017-08-05 by WereCatf

zador-blood-stained, 2017-08-05:
@WereCatf 
What was the problem exactly? if it's easy to encounter this situation maybe we could add extra checks to the build script.