armbian/build #729: Is zram no longer valid?

kometchtech, 2017-08-11:
In the past forum, it is written that zram module is enabled, but was this officially not effective? (It's written on the forum as a test)

https://forum.armbian.com/index.php?/topic/322-request-zram-kernel-module-available/
https://forum.armbian.com/index.php?/topic/4587-is-the-zram-module-not-valid/

Is it not effective even in the current version?

```
kometch@dns02:~$ uname -r
4.12.4-odroidc2
```
```
kometch@dns02:~$ sudo modprobe -Rf zram
modprobe: FATAL: Module zram not found in directory /lib/modules/4.12.4-odroidc2
```

Best regards.

igorpecovnik, 2017-08-11:
C2 mainline kernel is under development. Feel free to add missing things.

kometchtech, 2017-08-12:
Thank you for answering.

Since the mainline image in odroid c2 is handled during development for the time being, as of now the image prepared by the official is not validated this time.
Is that you need to run build.sh yourself, create it and activate it?

igorpecovnik, 2017-08-12:
`Is that you need to run build.sh yourself, create it and activate it?`

You can also edit config file and wait for next automated build (currently it is on hold for other reasons): https://github.com/armbian/build/blob/master/config/kernel/linux-odroidc2-next.config

commit b868c262 (2017-09-06): Add ZRAM to odroidc2-next config
Closes #729
closed 2017-09-06 by zador-blood-stained