armbian/build #730: Missing linux-heders on compiled image

nintendontit, 2017-08-12:
I'm compiling image for H3 devices (sun8i)
From yesterday on compiled image on directory /usr/src/ i can't find the linux-heders folder for compile modules and drivers that require it.
Why? Is an update issue?

igorpecovnik, 2017-08-12:
We plan not to install them by default. If you want them, build with INSTALL_HEADERS="yes"

nintendontit, 2017-08-12:
Ok thank you.
Need to close? Or can leave here in waiting you update the compilation guide?

zador-blood-stained, 2017-08-12:
It's already in the [build options docs](https://docs.armbian.com/Developer-Guide_Build-Options/) (though it needs to be moved to the "Advanced" section), and eventually it should be reflected in the changelog once we start preparing it.

closed 2017-08-16 by nintendontit