armbian/build #731: u-boot deb postinstall script creates /dev/mmcblk0

h0bbyte, 2017-08-20:
Hi.
I use OPi PC2 with u-boot on SPI NOR Flash and not use TF.
```
$ cat /proc/cmdline 
root=UUID=ab84ee7e-2b52-4eeb-830c-a8d99481ce22 rootwait rootfstype=ext4 console=tty1 console=ttyS0,115200 panic=10 consoleblank=0 loglevel=1 ubootpart=   cgroup_enable=memory swapaccount=1
```
```
hobbyte@orangepipc2:~$ ll /dev/mmc*
ls: cannot access '/dev/mmc*': No such file or directory
```
```
hobbyte@orangepipc2:~$ sudo apt-get install ./linux-u-boot-dev-orangepipc2_5.32.170820_arm64.deb --reinstall
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'linux-u-boot-orangepipc2-dev' instead of './linux-u-boot-dev-orangepipc2_5.32.170820_arm64.deb'
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/hobbyte/linux-u-boot-dev-orangepipc2_5.32.170820_arm64.deb linux-u-boot-orangepipc2-dev arm64 5.32.170820 [197 kB]
(Reading database ... 139692 files and directories currently installed.)
Preparing to unpack .../linux-u-boot-dev-orangepipc2_5.32.170820_arm64.deb ...
Unpacking linux-u-boot-orangepipc2-dev (5.32.170820) over (5.32.170820) ...
Setting up linux-u-boot-orangepipc2-dev (5.32.170820) ...
Updating u-boot on device /dev/mmcblk0
```
```
hobbyte@orangepipc2:~$ ll /dev/mmc*
-rw-r--r-- 1 root root 505540 Aug 20 16:28 /dev/mmcblk0
```

zador-blood-stained, 2017-08-20:
Should be partially fixed by 78bac39. Unattended or partially automated updating on SPI flash will take some extra time to implement, and it's a low priority task for now.

h0bbyte, 2017-08-20:
tnx for a workaround.

closed 2021-06-14 by lanefu