armbian/build #737: asus tinkerboard boot failure

flomo29, 2017-08-24:
Hi.
Before building the kernel for "board: tinkerboard, branch: next, distr.: xenial" and everything else are left on default, there was a failure applying the "packaging-4.x-NEXT-with-postinstall-scripts.patch".
But building the armbian image run successfully.

When i boot the tinkerboard, the boot process stuck in the middle, before activation of the ethernet part. I thing it could be an error in this file, because of the large "dtb"-part.

Tonymac32, 2017-08-24:
I will look into this later (not at my desk now), do you have console output from the UART2 you could share?

Tonymac32, 2017-08-25:
4.11 to 12 changed the postinstall scripts enough to reject the patch.  I've tested it,

https://github.com/armbian/build/commit/7e07d7b194e8c0eb84347c68ed186252e475c245

Try to build now, if successful on your side we can close this.

mpmc, 2017-08-25:
@Tonymac32, I've said this already on the forums, but I'll say it again here, thanks for getting Armbian working on the Tinkerboard.

Here's the log from the build process

```
Displaying message: Command line: setting RELEASE to stretch info
Displaying message: Preparing host info
Displaying message: Build host OS release xenial info
Displaying message: Syncing clock host info
## BUILD SCRIPT ENVIRONMENT

Version: 4f08c97d745892b6d664b96f8a4f84f48ee1f53f

## BUILD CONFIGURATION

Build target:
Board: tinkerboard
Branch: next
Desktop: no

Kernel configuration:
Repository: git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git
Branch: branch:linux-4.12.y
Config file: linux-rockchip-next

U-boot configuration:
Repository: git://git.denx.de/u-boot.git
Branch: branch:master
Config file: tinker-rk3288_defconfig

Partitioning configuration:
Root partition type: ext4
Boot partition type: (none)
User provided boot partition size: 0
Offset: 1

CPU configuration:
600000 - 1900000 with conservative
Displaying message: Downloading sources  info
Displaying message: Checking git sources u-boot master info
Displaying message: Creating local copy
Displaying message: Fetching updates
Displaying message: Checking out
Displaying message: Checking git sources linux-mainline linux-4.12.y info
Displaying message: Fetching updates
Displaying message: Checking out
Displaying message: Checking git sources sunxi-tools master info
Displaying message: Up to date
Displaying message: Checking git sources armbian-config dev info
Displaying message: Fetching updates
Displaying message: Checking out
Displaying message: Checking git sources rkbin-tools master info
Displaying message: Up to date
Displaying message: Cleaning output/debs for tinkerboard next info
Displaying message: Cleaning u-boot/master info
Displaying message: Compiling u-boot 2017.09-rc2 info
Displaying message: Compiler version arm-linux-gnueabihf-gcc 6.3.1 info
Displaying message: Checking out sources
Displaying message: Cleaning u-boot/master info
Displaying message: Started patching process for u-boot rockchip-tinkerboard-next info
Displaying message: Looking for user patches in userpatches/u-boot/u-boot-rockchip info
Displaying message: Building deb linux-u-boot-next-tinkerboard_5.32_armhf.deb info
Displaying message: Cleaning linux-mainline/linux-4.12.y info
Displaying message: Patching kernel for compiler support
Reversed (or previously applied) patch detected!  Assuming -R.
Displaying message: Started patching process for kernel rockchip-next info
Displaying message: Looking for user patches in userpatches/kernel/rockchip-next info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] 0000-mali-r16p0-00rel0.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] 0001-ARM-dts-rockchip-fix-the-regulator-s-voltage-range-o.patch succeeded info
Displaying message: ... [\e[32ml\e[0m][\e[32mc\e[0m] 0001-Readaptation-of-Rockchip-DRM-patches-provided-by-ARM.patch succeeded info
[... 110 lines cut ...]
   38 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 crypto                                                                                                                                         
   39 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 kintegrityd                                                                                                                                    
   40 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 bioset                                                                                                                                         
   41 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 kblockd                                                                                                                                        
   42 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 devfreq_wq                                                                                                                                     
   43 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 cfg80211                                                                                                                                       
   44 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 rpciod                                                                                                                                         
   45 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 xprtiod                                                                                                                                        
   46 root      20   0       0      0      0 S   0.0  0.0   0:00.66 kworker/2:1                                                                                                                                    
   47 root      20   0       0      0      0 S   0.0  0.0   0:00.00 kswapd0                                                                                                                                        
   48 root       0 -20       0      0      0 S   0.0  0.0   0:00.00 bioset                                                                                                                                         
   49 root      20   0       0      0      0 S   0.0  0.0   0:00.00 ecryptfs-kthrea                           
```
[... cut ...]
19:30:58:  600MHz  1.21  16%   4%   9%   0%   1%   0% 33.2°C
19:31:03:  600MHz  1.27  16%   4%   9%   0%   1%   0% 32.7°C
19:31:08:  600MHz  1.25  16%   4%   9%   0%   1%   0% 32.7°C
19:31:13:  600MHz  1.29  16%   4%   9%   0%   1%   0% 33.2°C
19:31:18:  600MHz  1.34  16%   4%   9%   0%   1%   0% 33.2°C
19:31:23:  600MHz  1.24  16%   4%   9%   0%   1%   0% 33.6°C^C
```

If you want me to Test anything let me know, I'm more than happy to test anything if required as I won't be using this board fully until it's fully stable under armbian :).

Tonymac32, 2017-08-25:
The 5V on the board never shuts down, so that will always be the case.  The processor and derived voltages shut down, but not the main 5V.

For the wifi, does it still do that on reboot?  It looks like your hardware is not powering up.  I'll check my build from last night, but I did not notice anything out of the ordinary.  Can you see what Tinker OS does on your particular board?

Tonymac32, 2017-08-25:
Spoke too soon, I see it entering/leaving powersave in the log.  Did you set a script of some kind that needs the interface?  I haven't done a console-only build, I'll have to do that to see if the management is buggy.  If I confirm I'd say open a different issue, we can close this one if the build error is fixed, waiting for confirmation from @flomo29

mpmc, 2017-08-25:
@Tonymac32 The only thing I added was the xbox tuner patches as shown in the log, nothing else. :) will open another issue with the armbian log link attached.

Yeah I shouldn't have hijacked his issue, my apologies!

Tonymac32, 2017-08-25:
No worries, it at least showed the build system appears to be working as far as build --> boot goes.

Tonymac32, 2017-08-27:
I've built 5 images since the patch, and @mpmc has built a few as well, I am closing this.

closed 2017-08-27 by Tonymac32