armbian/build #741: Tinkerboard: Wifi dmesg spam

mpmc, 2017-08-25:
As requested by @Tonymac32, please see [here](https://github.com/armbian/build/issues/737#issuecomment-325016192)

But to summarise, dmesg is being spammed with messages about wifi and NetworkManager is keeping the CPU busy. Apart from the xbox tuner patches I added myself to get the xbox tuner running in Armbian I haven't changed anything. 

Please see the above link for the armbian log :)

Tonymac32, 2017-08-25:
No issue on 4.12.9 Xenial Desktop, I'll build a sever image and see.

Tonymac32, 2017-08-26:
4.12.9 Xenial Server, I get 4 link not ready messages from IPv6.  1 from eth0, 3 from wlan0, then it goes into its typical powersave cycle.

Processor is sitting idle.

Try building without your patches and see what the result is.

mpmc, 2017-08-26:
@Tonymac32 rebuilt the kernel without the xbox tuner patches & the result is the same.. Could it be because I'm using Stretch do you think?

```
[    9.319990] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  619.620443] r8723bs: module is from the staging directory, the quality is unknown, you have been warned.
[  619.629221] RTL8723BS: module init start
[  619.629232] RTL8723BS: rtl8723bs v4.3.5.5_12290.20140916_BTCOEX20140507-4E40
[  619.629238] RTL8723BS: rtl8723bs BT-Coex version = BTCOEX20140507-4E40
[  619.629741] pnetdev = ed31c800
[  619.722786] RTL8723BS: rtw_ndev_init(wlan0)
[  619.731959] RTL8723BS: module init ret =0
[  619.799269] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  619.814873] rtl8723bs: acquire FW from file:rtlwifi/rtl8723bs_nic.bin
[  621.127986] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  621.233740] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  621.287060] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  621.393552] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  621.518402] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  621.632943] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  621.753433] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  621.869676] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  622.007743] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  622.132917] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  622.253285] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  622.378348] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  622.501978] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  622.621738] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  622.746869] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  622.772493] RTL8723BS: nolinked power save enter
[  622.911878] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  623.154534] RTL8723BS: nolinked power save leave
[  623.476425] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  623.619853] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  623.760482] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  623.891459] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  624.020898] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  624.161808] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  624.296990] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  624.429176] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  624.559314] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  624.689191] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  624.820388] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  624.970886] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  625.097046] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  625.234618] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  625.365853] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  625.497319] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  625.630375] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  625.763384] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  625.897843] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  626.047511] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  626.195765] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  626.211646] RTL8723BS: nolinked power save enter
[  626.474239] RTL8723BS: nolinked power save leave
[  628.047799] RTL8723BS: nolinked power save enter
[  628.173060] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  628.410262] RTL8723BS: nolinked power save leave
[  629.981594] RTL8723BS: nolinked power save enter
[  630.108408] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  630.354553] RTL8723BS: nolinked power save leave
[  631.932740] RTL8723BS: nolinked power save enter
[  632.069014] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  632.306447] RTL8723BS: nolinked power save leave
[  633.884969] RTL8723BS: nolinked power save enter
[  634.016942] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  634.258246] RTL8723BS: nolinked power save leave
[  635.836753] RTL8723BS: nolinked power save enter
[  635.970475] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  636.210432] RTL8723BS: nolinked power save leave
[  636.528346] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[... 245 lines cut ...]
[  679.822633] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  679.953762] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  680.093213] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  680.232163] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  680.370859] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  680.503545] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  680.634710] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  680.767083] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  680.898794] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  681.030062] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  681.169979] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  681.299118] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  681.430998] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  681.564519] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  681.697329] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  681.827195] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  681.957787] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  682.088339] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  682.151412] RTL8723BS: module exit start
[  682.159188] RTL8723BS: rtw_ndev_uninit(wlan0)
[  682.198962] RTL8723BS: rtw_cmd_thread: DriverStopped(1) SurpriseRemoved(0) break at line 481
[  682.199466] RTL8723BS: rtw_dev_unload: driver not in IPS
[  682.219705] RTL8723BS: module exit success
```
This is after I modprobed & then rmmod

Tonymac32, 2017-08-26:
Ah.  Could be, I'll try that.  We don't have stretch worked out yet

Tonymac32, 2017-08-27:
OK, it is something to do with Stretch, built a clean image and got the same mess you described.  Since Stretch isn't officially supported yet (it can only be built with "expert" settings enabled for that reason), I can only recommend you use Ubuntu Xenial for now.

mpmc, 2017-09-17:
@Tonymac32 

Using xenial as recommended to avoid the spam, but now I'm getting power save enter/leave spam instead every so often. I'm not 100% sure this is related to the wifi, but decided to ask here instead of opening another issue (I assume it is due to "RTL8723BS").

dmesg -> http://termbin.com/qxi5

johnobe, 2017-09-26:
@Tonymac32, @mpmc 

I googled around for a fix for this tonight and found this link: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=839605

There's a proposed solution to add the following lines to `/etc/NetworkManager/NetworkManager.conf`:

```
[device-mac-randomization]
wifi.scan-rand-mac-address=no
```

This seemed to fix the spam for me and my wifi is working.

mpmc, 2017-10-05:
@johnobe Thanks for the fix, I'll look into testing it out on a reinstall.

zador-blood-stained, 2017-10-24:
Should be fixed by https://github.com/armbian/build/commit/46e6c336639bcfc0c69518e0c3d2e93a69b96418

closed 2017-10-24 by zador-blood-stained

mpmc, 2017-10-29:
@zador-blood-stained It has fixed the "IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready" but I'm still getting ```RTL8723BS: nolinked power save enter```. spam on Stretch. full log [here](http://sprunge.us/FNTO).

zador-blood-stained, 2017-11-03:
8723BS is still a staging driver so it will need some fixes including hiding/removing debugging messages by default (similar to this https://github.com/Icenowy/rtl8723cs/commit/3fed0ea2451c513b826a83e0c4b6bb84bf1ec34a)

mpmc, 2017-11-03:
@zador-blood-stained Thanks for the info. A fix can be waited for  :+1: My reply was more of just letting people know there's still an issue :)