armbian/build #744: Enable CMA back for sun7i kernel

iavael, 2017-08-26:
Can you enable CMA back, please, so that usage of libump won't cause kernel oops and Xorg crash? I see, that patch with CMA-enabled UMP is applied, but CONFIG_CMA is not set.

```
CONFIG_CMA=y
# CONFIG_CMA_DEBUG is not set
#
# Default contiguous memory area size:
#
CONFIG_CMA_SIZE_MBYTES=192
CONFIG_CMA_SIZE_SEL_MBYTES=y
# CONFIG_CMA_SIZE_SEL_PERCENTAGE is not set
# CONFIG_CMA_SIZE_SEL_MIN is not set
# CONFIG_CMA_SIZE_SEL_MAX is not set
CONFIG_CMA_ALIGNMENT=8
CONFIG_CMA_AREAS=7
```
(taken from https://forum.kodi.tv/showthread.php?tid=254202)

zador-blood-stained, 2017-09-06:
Default CMA size is set to 16M, can be changed by using the `cma` kernel parameter, i.e. on recent enough images by adding 
```
extraargs=cma=192M
```
to `/boot/armbianEnv.txt`

closed 2017-09-06 by zador-blood-stained

PiPoly, 2017-10-13:
Sorry to dig up this subject but with my cubietruck I have an issue with the CMA.
With the last build, debian server version, legacy kernel, the board start and works very well with cma at default value (16M) but if I change cma kernel parameter by adding `extraargs=cma=192M` to /boot/armbianEnv.txt and I reboot. Then my board doesn't start, it's just display the bootloader initialisation and it's reboot.

iavael, 2017-10-15:
@PiPoly cma=96M is enough to activate GLES and VDPAU.

PiPoly, 2017-10-15:
ok big thanks @iavael it's working with the configuration `extraargs=cma=96M` 
But I don't understand why it's doesn't work with 128M or 192M. If somewone can explain me.
Ps: I work with a cubietruck.

iavael, 2017-10-15:
There's a hardcoded limit for CMA size.

This string gives a hint
```
UMP<2>: Using CMA memory backend, allocation limit: 134217728
```

PiPoly, 2018-03-20:
Hello, I can't change CMA more than 100M after that my board doesn't start and reboot in loop. 
I need to increase the cma to 128M at least. 
It was working with the version `Armbian_5.11_Cubietruck_Ubuntu_trusty_3.4.112`
Where is the difference between this version and now?
Thanks

commit 13003033 (2017-09-06): Enable CMA for sun7i-default kernel
Closes #744