armbian/build #760: odroidc2-next: no ethernet after boot

flomo29, 2017-09-08:
I build the odroidc2 image with the next branch after you change the odroidc2 next kernel config in the last commits. After powering on the board, there is no ethernet available!

igorpecovnik, 2017-09-08:
I made [this image](https://dl.armbian.com/odroidc2/nightly/Armbian_5.32.170909_Odroidc2_Ubuntu_xenial_next_4.12.11.7z) a few hours ago. Ethernet is working, logs: http://sprunge.us/PWVQ

flomo29, 2017-09-09:
Okay. I tried different settings and different cards. When i build the image with 1 ext4 partition (default setting) everything works fine! But when i choose btrfs for root, i got problems with the ethernet. it cycles on and off every few seconds. when i choose the emmc card with btrfs, ethernet is completly not working. I see on my power measurement unit, that the current moves between idle 370 mA and 470 mA, up and down. For me it looks like a loop cycle or something else.

igorpecovnik, 2017-09-10:
No idea what could cause this. There were some recent changes to BTRFS part, but only for auto-expanding part. I also experienced a problem with BTRFS on the same (4.13) kernel but on imx6 platform (Udoo). But when using nand-sata-install utility, not with BTRFS image from scratch. Perhaps this is related? ... while it works perfectly fine on Banana Pi and kernel 4.13

ThomasKaiser, 2017-09-11:
Maybe the boot script is not ready for btrfs since with rootfs on btrfs we have two partitions and the check for `.next` will fail so the wrong DT is loaded then. See ODROID XU4: https://github.com/armbian/build/commit/8d4083c82f0f6ed644b032223479f524a0bbdd79

ThomasKaiser, 2017-09-12:
Quick check with https://pastebin.com/kw70R4EJ worked for me. I would call this rather inelegant  but have also to admit that I'm an u-boot noob.

    diff --git a/config/bootscripts/boot-odroid-c2.ini b/config/bootscripts/boot-odroid-c2.ini
    index dd571d0..8731388 100644
    --- a/config/bootscripts/boot-odroid-c2.ini
    +++ b/config/bootscripts/boot-odroid-c2.ini
    @@ -148,7 +148,7 @@ setenv maxcpus "4"
     setenv max_freq "1536"  # 1.536GHz
     
     # Default Console Device Setting
    -if test -e mmc 0:1 boot/.next; then setenv condev "console=ttyAML0,115200n8"; else setenv condev "console=ttyS0,115200n8 console=tty0"; fi
    +if test -e mmc 0:1 boot/.next -o -e mmc 0:1 .next ; then setenv condev "console=ttyAML0,115200n8"; else setenv condev "console=ttyS0,115200n8 console=tty0"; fi
     
     ###########################################
     setenv verbosity "1"
    @@ -157,7 +157,7 @@ setenv verbosity "1"
     setenv bootargs "root=${rootdev} rootwait rootflags=data=writeback rw rootfstype=${rootfstype} ${condev} no_console_suspend consoleblank=0 hdmimode=${m} m_bpp=${m_bpp} vout=${vout} fsck.repair=yes loglevel=${verbosity} net.ifnames=0 ${extraargs}"
     
     # Mainline boot arguments
    -if test -e mmc 0:1 boot/.next; then setenv bootargs "root=${rootdev} rootwait rootflags=data=writeback rw rootfstype=${rootfstype} ${condev} no_console_suspend consoleblank=0 fsck.repair=yes loglevel=${verbosity} net.ifnames=0 ${extraargs}"; fi
    +if test -e mmc 0:1 boot/.next -o -e mmc 0:1 .next ; then setenv bootargs "root=${rootdev} rootwait rootflags=data=writeback rw rootfstype=${rootfstype} ${condev} no_console_suspend consoleblank=0 fsck.repair=yes loglevel=${verbosity} net.ifnames=0 ${extraargs}"; fi
     
     # Booting
     setenv initrd_high "0x60000000"
    @@ -176,5 +176,7 @@ if test "${mesontimer}" = "1"; then fdt rm /timer; fi
     if test "${nographics}" = "1"; then fdt rm /meson-fb; fdt rm /amhdmitx; fdt rm /picdec; fdt rm /ppmgr; fdt rm /meson-vout; fdt rm /mesonstream; fdt rm /deinterlace; fdt rm /codec_mm; fdt rm /reserved-memory; fdt rm /aocec; fi
     
     if test -e mmc 0:1 boot/.next; then ext4load mmc 0:1 ${dtb_loadaddr} boot/dtb/amlogic/meson-gxbb-odroidc2.dtb; fi
    +if test -e mmc 0:1 .next; then ext4load mmc 0:1 ${dtb_loadaddr} dtb/amlogic/meson-gxbb-odroidc2.dtb; fi
     if test -e mmc 0:1 boot/.next; then ext4load mmc 0:1 0x01080000 boot/uImage; fi
    -if test -e mmc 0:1 boot/.next; then bootm 0x01080000 ${initrd_loadaddr} ${dtb_loadaddr}; else booti ${loadaddr} ${initrd_loadaddr} ${dtb_loadaddr}; fi
    +if test -e mmc 0:1 .next; then ext4load mmc 0:1 0x01080000 uImage; fi
    +if test -e mmc 0:1 boot/.next -o -e mmc 0:1 .next; then bootm 0x01080000 ${initrd_loadaddr} ${dtb_loadaddr}; else booti ${loadaddr} ${initrd_loadaddr} ${dtb_loadaddr}; fi

ThomasKaiser, 2017-09-12:
No, doesn't work for me. I adjusted bootscript on an SD card modified with `nand-sata-install` before. Booted once but doesn't boot anymore. No idea.

zador-blood-stained, 2017-10-27:
Since C2 next and dev branches were updated to use the mainline u-boot I built a dev btrfs image (since next is currently broken due to other reasons) and it works fine, including Ethernet and auto FS resize: http://sprunge.us/OSCF

So if there are no other issues (like nand-sata-install) this can be closed.

zador-blood-stained, 2017-10-28:
OK, nand-sata-install looks broken. It used wrong FS type (ext4) and incompatible parameters (noatime,nodiratime,commit,...) for /media/mmcboot - though I installed from btrfs SD to btrfs USB.

zador-blood-stained, 2017-10-28:
And mounting the second partition is wrong anyway - it should've used existing /boot partition

igorpecovnik, 2017-10-29:
I just installed PC2 latest stretch desktop image to USB drive. It looks ok:
```
Filesystem     Type      Size  Used Avail Use% Mounted on
udev           devtmpfs  431M     0  431M   0% /dev
tmpfs          tmpfs     100M  3.2M   97M   4% /run
/dev/sda1      btrfs      29G  1.2G   26G   5% /
tmpfs          tmpfs     498M     0  498M   0% /dev/shm
tmpfs          tmpfs     5.0M  4.0K  5.0M   1% /run/lock
tmpfs          tmpfs     498M     0  498M   0% /sys/fs/cgroup
tmpfs          tmpfs     498M  8.0K  498M   1% /tmp
/dev/mmcblk0p1 ext4       15G  2.4G   12G  17% /media/mmcboot
log2ram        tmpfs      50M  4.2M   46M   9% /var/log
tmpfs          tmpfs     100M  8.0K  100M   1% /run/user/1000
tmpfs          tmpfs     100M     0  100M   0% /run/user/0
```
http://sprunge.us/VPaU 

Huh?

zador-blood-stained, 2017-10-29:
> I just installed PC2 latest stretch desktop image to USB drive.

Did your initial image have btrfs root too? It should work with default ext4 images, but with 2 partition images (separate /boot) it needs some special logic to copy /boot entry from old fstab to new and to mount the SD rootfs with an appropriate FS (also we won't need a bind mount entry in this case)

igorpecovnik, 2017-10-29:
No, it was a normal ext4 image.

closed 2018-01-20 by zador-blood-stained