armbian/build #761: Suggestion: Make user-defined kernel .config naming more flexible?

g40, 2017-09-11:
I am wondering if the test for a user defined kernel .config in userpatches could not include the board name so as to enable multiple projects to be maintained in parallel? 

For example I'm trying to establish why USB gadget works with the dev kernel on an OPI One but not on a NanoPi MI. The build system is quite happy to maintain 2 images and even 2 kernel source directories, so it's slightly tiresome to have to keep juggling the relevant .config files.

i.e. 

`[[ -z $LINUXCONFIG ]] && LINUXCONFIG="linux-${LINUXFAMILY}-${BRANCH}"`

becomes

`[[ -z $LINUXCONFIG ]] && LINUXCONFIG="linux-${BOARDNAME}-${BRANCH}"`

It would be even better to be able to use $KERNELBRANCH in the name but that makes things difficult for Windows due to the embedded ':' ...
 
TAIA

Jerry.

zador-blood-stained, 2017-09-11:
> For example I'm trying to establish why USB gadget works with the dev kernel on an OPI One but not on a NanoPi MI.

Why? If you don't want the driver to be probed/loaded you should tweak the DT, you don't need to disable/remove any kernel modules.

> The build system is quite happy to maintain 2 images and even 2 kernel source directories, so it's slightly tiresome to have to keep juggling the relevant .config files.

But it won't be exactly happy to maintain 2 kernel package sets compiled with a different configuration.

And adding this
```
[[ -z $LINUXCONFIG ]] && LINUXCONFIG="linux-${BOARDNAME}-${BRANCH}"
```
to the production setup means that even if initial images are compiled against different board-specific configs, all board-specific changes will be lost on upgrade because the repository van contain only one copy of `linux-image-dev-sun8i`.

zador-blood-stained, 2017-09-11:
Also you can always tweak LINUXCONFIG in `userpatches/lib.config` for yourself and implement any board, branch and SoC family specific logic that you need.

g40, 2017-09-11:
@zador-blood-stained I had not thought about tweaking LINUXCONFIG. Let me see how that works out. Thanks for listening.

g40, 2017-09-15:
@zador-blood-stained Yes, very good call, thank you. Setting the kernel config via lib.config is the way to go.

closed 2017-09-15 by g40