armbian/build #762: Building with ./compile.sh fails in path with spaces

kopegit, 2017-09-13:
... with error:

root@ubuntu:~/H5 kernel/build# ./compile.sh
Error: missing build directory structure
Please clone the full repository https://github.com/armbian/build/

As I see, $SRC in the script is used without "", e.g.: cd $SRC

igorpecovnik, 2017-09-13:
Can you be more precise? I noticed some troubles with u-boot compilation this morning on Allwinner boards, while I was not able to nail the bug down yet.

ThomasKaiser, 2017-09-13:
> Can you be more precise?

His `cwd` contains a space ('~/H5 kernel') and since almost all shell script code contained in the build system does **not** take care of correct variable quoting it should be considered a feature that `compile.sh` fails that early.

zador-blood-stained, 2017-09-13:
We should put this requirement (no spaces in the full path to the build script location) in the main README and maybe in the "Build preparation" documentation section.

zador-blood-stained, 2017-10-09:
Documentation fixed in https://github.com/armbian/build/commit/66d9c4ad86458404bd62448ffdbc5053215d63fd and https://github.com/armbian/documentation/commit/66f17107c4a38f8bb2b0d6913ece188a320382f9

closed 2017-10-09 by zador-blood-stained