armbian/build #763: odroid-c2 next: no cpufreq driver installed

flomo29, 2017-09-13:
When i run the armbian odroid-c2 next image, the /sys/devices/system/cpu/cpu0/cpufreq folder is unavailable. This belongs to some other boards with the next branch too.
> root@odroidc2:~# cpufreq-info
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 1:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 2:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 3:
  no or unknown cpufreq driver is active on this 
  maximum transition latency: 4294.55 ms.

flomo29, 2017-09-13:
> root@odroidc2:~# cat /boot/config-4.12.12-odroidc2 | grep FREQ
CONFIG_CPU_FREQ=y
CONFIG_CPU_FREQ_GOV_ATTR_SET=y
CONFIG_CPU_FREQ_GOV_COMMON=y
#CONFIG_CPU_FREQ_STAT is not set
#CONFIG_CPU_FREQ_DEFAULT_GOV_PERFORMANCE is not set
#CONFIG_CPU_FREQ_DEFAULT_GOV_POWERSAVE is not set
#CONFIG_CPU_FREQ_DEFAULT_GOV_USERSPACE is not set
CONFIG_CPU_FREQ_DEFAULT_GOV_ONDEMAND=y
#CONFIG_CPU_FREQ_DEFAULT_GOV_CONSERVATIVE is not set
#CONFIG_CPU_FREQ_DEFAULT_GOV_SCHEDUTIL is not set
CONFIG_CPU_FREQ_GOV_PERFORMANCE=y
CONFIG_CPU_FREQ_GOV_POWERSAVE=m
CONFIG_CPU_FREQ_GOV_USERSPACE=m
CONFIG_CPU_FREQ_GOV_ONDEMAND=y
CONFIG_CPU_FREQ_GOV_CONSERVATIVE=m
CONFIG_CPU_FREQ_GOV_SCHEDUTIL=y
CONFIG_CPUFREQ_DT=y
CONFIG_CPUFREQ_DT_PLATDEV=y
#CONFIG_ARM_BIG_LITTLE_CPUFREQ is not set
#CONFIG_ARM_DB8500_CPUFREQ is not set
#CONFIG_ARM_KIRKWOOD_CPUFREQ is not set
#CONFIG_ACPI_CPPC_CPUFREQ is not set
#CONFIG_QORIQ_CPUFREQ is not set
#CONFIG_PM_DEVFREQ is not set

flomo29, 2017-09-13:
I also want to change the cpu freq with cpu_freq var in the boot.ini file. i added the var in bootargs var, but nothing happend.

igorpecovnik, 2017-09-13:
http://linux-meson.com/doku.php and put dvfs into page search

zador-blood-stained, 2017-10-16:
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/arch/arm64/boot/dts/amlogic/meson-gxbb-odroidc2.dts?id=f7bcd4b6f6983d668b057dc166799716690423a4

zador-blood-stained, 2017-10-17:
If Hardkernel can provide patched firmware which doesn't report operating points higher than 1536MHz DVFS can be enabled back. Until then it is outside of our control and can't be fixed without possible reliability issues.

zador-blood-stained, 2017-10-17:
Looks like the updated blob was provided eventually, so we could try to reenable DVFS with it (and mainline u-boot) in the future
https://github.com/hardkernel/u-boot/commit/205c7b3259559283161703a1a200b787c2c445a5

commit 7c344dcf (2017-12-01): Reenable C2-next DVFS
Not tested since c2-next configuration is still broken
Closes #763
closed 2017-12-01 by zador-blood-stained