armbian/build #767: Enhancement? Pass kernel version string to customize-image.sh

g40, 2017-09-16:
Absolutely required for building and installing kernel modules and the like. Currently my script does 

`KVER=`(ls /lib/modules/*/modules.dep | cut -d'/' -f4)``

which is definitely a bit hit and miss.

zador-blood-stained, 2017-09-16:
I'm not sure what would be a good way to get the kernel version(s) to pass there.
You could get the kernel version like this:
```
KVER=$(ls -1 /lib/modules | tail -1)
```
or this:
```
KVER=$(ls -1 /var/lib/initramfs-tools | tail -1)
```

g40, 2017-09-19:
On a closely related note running `uname -r` in `customize-image.sh` returns the kernel version of the host, not the chrooted target. Is this expected or a bug?

zador-blood-stained, 2017-09-19:
> Is this expected or a bug?

Expected. It is only chroot, not full kernel virtualization.

closed 2017-09-19 by g40