armbian/build #768: Suggestion: option to use xconfig for kernel/u-boot?

g40, 2017-09-18:
Given that the only supported build host is a Ubuntu desktop there should not be any issues here. xconfig about x1000 times faster to find and navigate options.

Thanks for listening.

zador-blood-stained, 2017-09-18:
> Given that the only supported build host is a Ubuntu desktop

This is wrong (the desktop part), we support all kinds of headless/remote setups - including SSH, Docker, Vagrant, and there xconfig won't work.

g40, 2017-09-18:
Ah, fair point. I had completely overlooked the docker/ssh style setups. Would it not be possible to then have a config envvar somewhere? Default to menuconfig but can be overriden on command line or in lib.config?

g40, 2017-09-19:
@zador-blood-stained Are you interested in a PR for this one?

zador-blood-stained, 2017-09-19:
> Are you interested in a PR for this one?

I'll implement this when I find some free time. I just need to keep it simple and minimal.

g40, 2017-09-19:
Nice. Thanks.

closed 2017-09-19 by zador-blood-stained

ThomasKaiser, 2017-09-19:
You might send a PR against [armbian-documentation](https://github.com/armbian/documentation) mentioning how to set for example `KERNEL_MENUCONFIG=xconfig` and all the other yet undocumented stuff ;)

zador-blood-stained, 2017-09-19:
FYI xconfig may need additional build dependencies that are not included by default.

g40, 2017-09-20:
@ThomasKaiser yes. I am putting together a section that I will ask you guys to review if I may.

commit 195faaac (2017-09-19): Add an option for different kernel configuration menu
Closes #768