armbian/build #769: Lime2: UART3 overlay gives data on ttyS1

Timple, 2017-09-19:
I am using the latest self-compiled image for the Olimex Lime2: 

My armbianEnv looks like this:
```
$ cat /boot/armbianEnv.txt 
verbosity=1
logo=disabled
console=both
disp_mode=1920x1080p60
overlay_prefix=sun7i-a20
overlays=can pwm uart3
rootdev=UUID=1a7bf5b1-db74-405e-9b28-6575ca973f97
rootfstype=btrfs
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
```

However UART3 overlay gives data on ttyS1 while I expect data on ttyS3.
Changing uart3 to uart4 in armbianEnv.txt gives no output on any ttySX which suggests my wiring is indeed on uart3

Is this perhaps an overlay issue?

martinayotte, 2017-09-19:
The numbering is done by the kernel itself.
If you add "overlays=uart1 uart2 uart3", the uart3 will then be numbered as ttyS3.

zador-blood-stained, 2017-09-19:
The numbering should be controlled by aliases regardless of activated serial ports.

Timple, 2017-09-19:
I agree with @zador-blood-stained, otherwise adding a new serial port will require administration for unrelated programs / auxiliaries.

zador-blood-stained, 2017-09-19:
> I agree with @zador-blood-stained, otherwise adding a new serial port will require administration for unrelated programs / auxiliaries.

By "should" I meant that I expected it to work already, but I don't have a spare A20 based board to test what is the problem with existing overlays.

martinayotte, 2017-09-19:
I've some A20-Olimex, so I will give it a try later today ...

martinayotte, 2017-09-19:
Ok ! I've checked with one of my A20-Olinuxino, and enabled UART7 and it was properly numbered as /dev/ttyS7. This was test with kernel 4.13.2

zador-blood-stained, 2017-09-20:
I've found a typo/bug that affects only some overlays (including uart3). Will be fixed ASAP.

closed 2017-09-20 by zador-blood-stained

Timple, 2017-09-20:
Verified!

Thanks.

commit 3e2d1c0a (2017-09-20): Fix A10/A20 UART overlay aliases
Closes #769