armbian/build #770: Reboot results in poweroff on Olimex Lime2-emmc

Timple, 2017-09-20:
Moving [this topic](https://forum.armbian.com/index.php?/topic/4701-reboot-results-in-poweroff-on-olimex-lime2-emmc/) into an issue because it seems related to kernel and/or u-boot configuration / patches. 

### Issue
When I do a reboot the Olimex Lime2-emmc shuts down. No more information is shown on the display.
 
This happens with the latest Armbian_5.31_Lime2_Debian_jessie_next_4.11.5 image from the website.

For both EMMC as NAND memory 
 
It also happens with a self-build image at least with u-boot v2017.07 and v2017.05.

### Trials until this far:
 
#### Older Armbian_5.20_Lime2_Ubuntu_xenial_4.7.3 image: 
- rebooting works
- rebooting after unattended upgrade works
- rebooting after apt upgrade works once, no more thereafter
 
#### Older Armbian_5.24_Lime2_Ubuntu_xenial_next_4.9.5 image: 
- rebooting works
- apt-mark hold linux-u-boot-lime2-etc
- rebooting after apt upgrade works once, no more thereafter
 
#### Nighly Armbian_5.27.170614_Lime2_Ubuntu_xenial_next_4.11.4 image:
- rebooting does not work
 
#### Newest Armbian_5.31_Lime2_Ubuntu_xenial_next_4.11.5 image:
- rebooting does not work

### Und jetzt
At the moment I am compiling my images using docker. 
I have a testsetup at the moment, so any suggestions to try are welcome!

zador-blood-stained, 2017-09-20:
Most likely related:

https://groups.google.com/forum/#!searchin/linux-sunxi/reboot$20lime2|sort:relevance/linux-sunxi/EDvEsbHHqQI/SM5PPgbHCgAJ

https://groups.google.com/forum/#!searchin/linux-sunxi/reboot$20lime2|sort:relevance/linux-sunxi/nn-MNRIrR_k/Hzut1U4xDAAJ

But this issue most likely is not related to any Armbian specific patches or configuration tweaks, so it has to be fixed upstream, and without access to affected hardware (= Lime2 board samples) it will be hard to test the proposed patch by ourselves.

Timple, 2017-09-20:
Oef, this seems relevant indeed. 

I am trying to find the patches themselves, but I can only find the description of the patches in the mailing lists
```
Olliver Schinagl (8): 
  sunxi: board: Print error after power initialization fails 
  sunxi: pmic_bus: Decrease boot time by not writing duplicate data 
  power: axp209: Use BIT() macro 
  power: axp209: Define the chip version mask 
  power: axp209: Reduce magic values by adding defines for LDO[234] 
  power: axp209: Add support for voltage rate control on LDO3 
  power: axp209: Limit inrush current for broken boards 
  arm: sunxi: Enable inrush quirk on Olimex OLinuXino-A20-Lime2 

 arch/arm/mach-sunxi/pmic_bus.c        |  6 ++++ 
 board/sunxi/board.c                   | 14 ++++---- 
 configs/A20-OLinuXino-Lime2_defconfig |  1 + 
 drivers/power/Kconfig                 | 42 ++++++++++++++++++++++ 
 drivers/power/axp209.c                | 61 ++++++++++++++++++++++++++------ 
 include/axp209.h                      | 66 ++++++++++++++++++++++++++--------- 
 6 files changed, 158 insertions(+), 32 deletions(-) 
```

Might be me, I am not used to mailing lists.

The idea is to extract a patch here an apply it in the userpatches directory for armbian. That should work right?

zador-blood-stained, 2017-09-20:
> I am trying to find the patches themselves, but I can only find the description of the patches in the mailing lists

They are in the second thread that I linked, but you'll need to click the each "spoiler" to open it and then use the menu on the right to show the original message. Or you may have better luck finding the messages on the denx.de list archive: https://lists.denx.de/pipermail/u-boot/2017-March/282789.html

> The idea is to extract a patch here an apply it in the userpatches directory for armbian. That should work right?

Not exactly, patches may need manual rebasing on top of the latest sources.

Timple, 2017-09-25:
> Not exactly, patches may need manual rebasing on top of the latest sources.

I was expecting to modify things. It did not look very straightforward so I'll first look into older kernel / u-boot combinations. And try to pin-down the version (or maybe even commit) where it all went wrong.

fheslouin, 2017-09-27:
@Timple did you find a good workaround ? I'm also stuck with that bug...

fheslouin, 2017-09-27:
The only workaround I found so far is to remove declaration about `ldo3` in `sun7i-a20-olinuxino-lime2.dts`

Timple, 2017-10-05:
Well, due to happy private circumstances I won't be able to work on this for another two weeks.
To be continued... What does the workaround do if I may ask?

fheslouin, 2017-10-27:
@Timple this work around avoid the board to poweroff when we actually ask for a reboot

igorpecovnik, 2017-10-30:
Reopen in case this is not working for you or if you find a better way.

Timple, 2017-11-13:
The workaround works!
Thanks for implementing. This enables us to move forward again.

StefanMavrodiev, 2017-12-13:
This workaround breaks default branch. During booting when kernel initialize axp the boot stops.
```
<6>sunxi-rtc sunxi-rtc: rtc core: registered rtc as rtc0
<6>i2c /dev entries driver
config i2c gpio with gpio_config api
<6>axp_mfd 0-0034: AXP (CHIP ID: 0x41) detected
<6>axp_mfd 0-0034: AXP internal temperature monitoring enabled
<4>i2c i2c-0: Invalid probe address 0x00
<6>I2C: i2c-0: AW16XX I2C adapter
config i2c gpio with gpio_config api
<4>i2c i2c-1: Invalid probe address 0x00
<6>I2C: i2c-1: AW16XX I2C adapter
config i2c gpio with gpio_config api
<4>i2c i2c-2: Invalid probe address 0x00
<6>I2C: i2c-2: AW16XX I2C adapter
<5>cedar: failed to allocate memory buffer
[tt]----- avs_dev driver load... ----
[tt]-----      register iomem      ----
[tt]--- avs_dev driver load ok!! -----
[ace_drv] start!!!
[ace_drv] init end!!!
[pa_drv] start!!!
[pa_drv] init end!!!
<6>axp20_ldo1: 1300 mV 
<6>axp20_ldo2: 1800 <--> 3300 mV at 3000 mV 
�x

igorpecovnik, 2017-12-31:
Tested booting default and NEXT kernel on Lime A20

ZupoLlask, 2018-01-15:
@igorpecovnik Can you share NEXT linux-headers and linux-image for Lime A20, so I can also test that fix? I couldn't find it @ http://beta.armbian.com/pool/main/l/linux-upstream/

Thanks.

igorpecovnik, 2018-01-15:
https://beta.armbian.com/pool/main/l/linux-u-boot-lime2-next/

Timple, 2018-01-19:
Our issue seems to be back again after a dist-upgrade..

igorpecovnik, 2019-08-02:
It seems this is solved. Can't recreate this problem on my Lime 2 emmc (Rev K)

closed 2019-08-02 by igorpecovnik

Timple, 2019-08-02:
True, we haven't seen it in a while

rzr, 2020-09-09:
Is this problem present on current armbian's buster image ? 
my micro can't reboot and does not reply to ping once i run reboot command

igorpecovnik, 2020-09-09:
It was fixed, but ... I only have one version of Lime2 with eMMC, no Micro or other variants ... can't confirms or denies. In case you have troubles - you have to boot latest build (or update everything, namely u-boot) and provide logs (armbianmonitor -u) here https://forum.armbian.com/forum/7-allwinner-a20/

We tried to keep this build system bug tracker from board specific problems.

dx168b, 2021-02-03:
The same problem in Debian Booster image for Olimex "A20 micro" board.
After reboot:
```
U-Boot SPL 2020.10-armbian (Jan 05 2021 - 00:29:10 +0100)
DR@#$% (trash in console)
```
and system not booting. After pressing to PWR key - the system boot normally.

My board is A20 Olinuxino Micro 4G (with 4GB NAND memory) Rev. E

igorpecovnik, 2021-02-03:
This is build system bug tracker - please use this form: https://www.armbian.com/bugs to report to right place and required data.

dx168b, 2021-02-03:
okay

rzr, 2021-02-03:
Please also crosslink to:

-  https://forum.armbian.com/topic/12321-reboot-results-in-poweroff-on-olimex-lime2-emmc-and-olimex-micro/

dx168b, 2021-02-05:
I builded new u-boot image with next configs (copied from A20-OLinuXino-Lime2-eMMC_defconfig file):
```
...
CONFIG_AXP_ALDO3_VOLT=2800
CONFIG_AXP_ALDO3_VOLT_SLOPE_08=y
CONFIG_AXP_ALDO3_INRUSH_QUIRK=y
...
```
This is work fine.
The problem is solved.

mpastor, 2021-11-04:
I have the same problem on Olinuxino A20 Micro, u-boot 2021.10-rc4 (compiled from source) and Debian 11 (Bullseye). All I have to do to fix this problem is enable _ldo3 inrush quirk_ in u-boot:

```
# diff -u .config.default .config.fixed 
--- .config.default	2021-11-04 15:36:01.948831909 +0100
+++ .config.fixed	2021-11-04 15:33:56.215887904 +0100
@@ -1304,7 +1304,7 @@
 CONFIG_AXP_ALDO3_VOLT_SLOPE_NONE=y
 # CONFIG_AXP_ALDO3_VOLT_SLOPE_16 is not set
 # CONFIG_AXP_ALDO3_VOLT_SLOPE_08 is not set
-# CONFIG_AXP_ALDO3_INRUSH_QUIRK is not set
+CONFIG_AXP_ALDO3_INRUSH_QUIRK=y
 CONFIG_AXP_ALDO4_VOLT=2800
 # CONFIG_POWER_MT6323 is not set
 # CONFIG_DM_PWM is not set
```

rzr, 2023-12-17:
Has this quirk upstreamed to uboot ? I looks it's needed also for debian-12+

https://github.com/OLIMEX/OLINUXINO/issues/93#issuecomment-1859131100