armbian/build #773: ft6x drivers on legacy kernel

pooyasf, 2017-09-21:
Hi all,

I've been trying to install ft6x drivers on a nanopi neo air, which has armbian with legacy kernel, but it seems there is no support for ft6x capacitive touch panels. 
There is a code from Focaltech [here](https://github.com/focaltech-systems/drivers-input-touchscreen-FTS_driver), which is ft6x06 driver code. I added the code to kernel and compiled it, but the problem is: 

- what should I do with fex file? 
- Or is it right to do this? (I was not surprised that it didn't work).
- Can I change the current ft5x code to make ft6x works?

There is also a support for ft6x family on mainline kernel. Is there any hope to adopt that code?

Thanks.

igorpecovnik, 2017-09-21:
3rd party hardware is out of our support zone and know-how. Personally, I am not familiar with this hardware or if they are somehow compatible with referenced ones. Sorry.

zador-blood-stained, 2017-09-27:
https://forum.armbian.com/index.php?/topic/5227-ft6x-driver-for-legacy-kernel/&do=findComment&comment=39976

closed 2017-09-27 by zador-blood-stained