armbian/build #783: Enabling SPI Overlay prevents OPI Lite from booting (MAINLINE)

kutysam, 2017-09-23:
Do up a fresh build of armbian for OPI Lite (And possibly Others) --> MAINLINE

sudo armbian-config and enable spi-spidev overlay
reboot and it will boot loop forever
Console message
```
Applying kernel provided DT overlay sun8i-h3-spi-spidev.dtbo
385 bytes read in 395 ms (0 Bytes/s)
Applying kernel provided DT fixup script (sun8i-h3-fixup.scr)
## Executing script at 44000000
data abort
pc : [<5df7247c>]          lr : [<00000010>]
reloc pc : [<4a00447c>]    lr : [<ec092010>]
sp : 59f48ad8  ip : 0000003e     fp : 5dfc4805
r10: 59f5daa8  r9 : 59f4dee0     r8 : 00000001
r7 : 59f5dac1  r6 : 59f5dae0     r5 : 00000000  r4 : 5dfc4c04
r3 : 59f48af0  r2 : 00000031     r1 : 59f48af0  r0 : 0024f400
Flags: Nzcv  IRQs off  FIQs off  Mode SVC_32
Resetting CPU ...
resetting ...
```

kutysam, 2017-09-23:
Just saw https://forum.armbian.com/index.php?/topic/5221-orange-pi-one-bootlooping-with-current-armbian/

I'll close this if its fixed. Probably the image will come later on.

zador-blood-stained, 2017-09-24:
> I'll close this if its fixed. Probably the image will come later on.

Any updates on this? I was unable to reproduce the crash on a clean image, but it also didn't happen without adding additional stuff to armbianEnv.txt before the fix.

kutysam, 2017-09-26:
I'm sorry. Unable to test for weeks or months. I'll close this ticket first and if it works, I'll comment out. Highly probably that it is working after seeing the bugfix

closed 2017-09-26 by kutysam