armbian/build #785: wifi adapter

symbios24, 2017-09-27:
Hello, great os fast and accurate

i connected to internet using a wifi adapter, however when i tried to created an adhock/wifi network with a second wifi adapter i just cannot do it it says this error messages:

Connection failure 
Failed to add new connection 
Insufficient privileges

is this a common problem? because i tried with an older version and i got the same error.

i`m using a cubieboard 2

Thank you

igorpecovnik, 2017-09-27:
Thank you.

Please supply logs to see if the adapter is supported. type armbianmonitor -u with adapter in place.

symbios24, 2017-09-27:
i forgot the OS is ubuntu xenial desktop latest version i also tried this on my orange pi one board and got the same results, i will be back to you with the results soon,thanks

igorpecovnik, 2017-10-12:
Reopen when more data is provided.

closed 2017-10-12 by igorpecovnik

symbios24, 2017-10-17:
Hello, sorry for the delay my wifi card is an atheros ath9k_htc

igorpecovnik, 2017-10-17:
This one must work on Cubieboard with legacy kernel https://github.com/armbian/build/blob/master/config/kernel/linux-sun7i-default.config#L1536 and on most modern kernels, but only in beta/self-build kernels since I enable support for those adapters not long ago.

symbios24, 2017-10-17:
'thanks and how do i switch to that firmware?

igorpecovnik, 2017-10-17:
Login as root and type:

apt install armbian-firmware-full

symbios24, 2017-10-24:
i found the problem, the network manager needs to run as root and not as normal user,if you try to create a wifi network as normal user you cannot select any options everything is grayed out.if you run it as root everything is selectable,how can i run the network manager as root on every boot?

igorpecovnik, 2017-10-24:
This means you have some problem with desktop configuration. Here - at least with upstream Armbian desktop - works without needing root privileges. If you use some other desktop, you need to fix this. Please use forum for further communication: https://forum.armbian.com/index.php?/forum/25-peer-to-peer-technical-support/