armbian/build #789: U-Boot USB Keyboard Support

cztian, 2017-09-30:
The Armbian U-Boot does not support USB keyboard. In Orange Pi Prime, stdin is set to serial, not usbkbd. Please add USB keyboard support (CONFIG_USB_KEYBOARD) to U-Boot config file.

zador-blood-stained, 2017-10-01:
It is disabled on purpose because some USB devices send erroneous data which abort the default boot sequence. Serial console should be used instead.

zador-blood-stained, 2017-10-02:
We may reenable USB keyboards but switch to AUTOBOOT_KEYED + AUTOBOOT_KEYED_CTRLC instead of the defaults which stop the autoboot on any key press.

cztian, 2017-10-02:
Yes, if we can distinguish terminal interrupt from erroneous data.

zador-blood-stained, 2017-10-08:
@igorpecovnik, @ThomasKaiser, @martinayotte  - any objections against enabling AUTOBOOT_KEYED with  `<Ctrl-C>` and `<Esc><Esc>` as default sequences and enabling USB keyboard support back?

igorpecovnik, 2017-10-08:
No objections. Anyway, we will make many tests before release and can be disabled later if we found out it's not working properly.

martinayotte, 2017-10-08:
No objection either ...

ThomasKaiser, 2017-10-10:
Fine with that

cztian, 2017-10-17:
This file? https://github.com/armbian/build/blob/master/patch/u-boot/u-boot-sunxi/disable-usb-keyboards.patch

zador-blood-stained, 2017-10-17:
> This file? https://github.com/armbian/build/blob/master/patch/u-boot/u-boot-sunxi/disable-usb-keyboards.patch

Yes.
I prepared a patch for AUTOBOOT_KEYED but didn't have much time for properly testing it yet.

commit dd48a784 (2017-10-24): Enable USB keyboards in sunxi u-boot
Switch to AUTOBOOT_KEYED with <Space> and <Ctrl-C> as autoboot abort
methods
Increase sunxi bootdelay to 1
Closes #789
closed 2017-10-24 by zador-blood-stained