armbian/build #791: [tinkerboard] RTC DS1307

dragonlost, 2017-10-05:
Hello ! 

It's possible to add patch for activate RTC DS1307 at place of RTC RK808 on option of image construction ?

Tuto in french for activate DS1307 RTC : https://pludov.blogspot.fr/2017/10/asus-tinker-board-mettons-les-pendules.html

Tonymac32, 2017-10-10:
I don't think this fits into the build system proper, simply because it isn't part of the board out of the box. 

I'll leave it open for more input, but I think this is a generally "special case" situation.

zador-blood-stained, 2017-10-10:
Ideally support for devices like RTC should be implemented using overlays. Adding aliases to overlays can be used to set external RTC as rtc0 without disabling RK808 one in the kernel.

Tonymac32, 2017-10-10:
Yes, Rockchip recently started using overlays, through u-boot

Grabber, 2017-11-15:
@Tonymac32, not exactly on this thread, but may help:

```
# INSTALL i2c-tools
sudo apt-get install i2c-tools

# DETECT RTC module
sudo i2cdetect -y 0

#  CREATE i2c adapter for RTC module
echo ds3231 0x68 > /sys/class/i2c-adapter/i2c-0/new_device

# WRITE system date to RTC module
hwclock -w -f /dev/rtc1

# READ date from RTC module
hwclock -r -f /dev/rtc1

# READ date from RTC module and set system date
hwclock -s -f /dev/rtc1 
```

chwe17, 2019-09-29:
solved with: https://github.com/armbian/build/commit/2ed2e50e56a08129f5eecc869819be1d00d160bc#diff-c76dc06500d39355edb75806f42fd57c

closed 2019-09-29 by chwe17