armbian/build #793: Question: Installing kernel source to image?

g40, 2017-10-05:
Anything analogous to the INSTALL_HEADERS I've not been able to find?

I've tried adding `linux-source` to the `PACKAGE_LIST` var in config/lib but it does not want to install the .deb created by the build.

Any other workarounds come to mind? Hack `distributions.sh`?

If no other way, then wonder if this could be added as a very nice to have feature?

TAIA.

igorpecovnik, 2017-10-05:
Check build manual: https://docs.armbian.com/Developer-Guide_Build-Options/

zador-blood-stained, 2017-10-05:
Not sure if this is a really "nice" feature (source package is quite fat, it installs an archive that needs to be unpacked manually and it may have issues due to compiler version requirements), but we can add a new option, it shouldn't be more than 3-4 lines of code.

g40, 2017-10-06:
@zador-blood-stained Yes, that would be great. It is much quicker to do this host side rather than doing an SCP, untar, etc. etc on the target.

zador-blood-stained, 2017-10-08:
Using `INSTALL_KSRC=yes` option should install the kernel sources archive to `/usr/src/`. Unpacking this archive and copying a valid kernel configuration is up to the user (similar to standard Debian/Ubuntu where kernel sources are provided as an archive too).

closed 2017-10-08 by zador-blood-stained

g40, 2017-10-08:
Thanks. A great addition.