armbian/build #795: Orange Pi Win/Win+ no package linux-source for Orange Pi Win/Win+

mozg1986, 2017-10-06:
In the repository there is no package linux-source for Orange Pi Win/Win+.
While there are packages for other platforms.
In order to raise your core you want to deploy the compilation environment. If the repository was the package linux-source-x.x-x-sun50iw, it would be much easier.

igorpecovnik, 2017-10-06:
This is the source you are looking for: https://beta.armbian.com/pool/main/l/linux-source-4.13.4-sunxi64/

mozg1986, 2017-10-06:
It turns out there was a merger of distributions in one? Now try to install the kernel linux-image-next-sunxi64_5.33. See works whether Ethernet.

mozg1986, 2017-10-06:
With the new Board-run, but Ethernet doesn't work. Maybe I have something not installed or something is required to be edited manually? Where I can help?

igorpecovnik, 2017-10-06:
OrangePi win - if you build from sources with our tool - has ethernet support. Beware that this is the bleeding edge of the development process and things may work today and be broken tomorrow. Certainly, this kernel is not suitable for deployment.

zador-blood-stained, 2017-11-25:
Kernel source packages should be present in the repository or they will be created when building kernel packages using this build script.

> Where I can help?

Allwinner A64 (And OPi Win board specifically) has limited mainline kernel support. Please don't expect things to work if they are not listed as working [here](https://linux-sunxi.org/Linux_mainlining_effort#Status_Matrix) and please note that kernel patches come from the linux-sunxi community, we are mostly using patches created by other people.

closed 2017-11-25 by zador-blood-stained