armbian/build #798: Adding Device Tree Overlays on sunxi64-next platform

cztian, 2017-10-12:
The programs `armbian-config-jobs` and  `armbian-config-submenu` lack the code of detecting sunxi64 platform (the overlay files for sunxi64 are in `/boot/dtb/allwinner/overlays`, not `/boot/dtb/overlays`, and `LINUXFAMILY` should be `sun*i*`, not `sun*i`), causing A64/H5 boards cannot add overlays from `armbian-config` utility. Please add responding code for A64/H5 to these utilities.

Szalacinski, 2017-10-29:
I'm having the issue of the Orange Pi Zero Plus 2 H5 not detecting any USB controllers with the new mainline kernel images.  Is this why?

igorpecovnik, 2017-10-29:
Only existing hardware is enabled by default. https://docs.armbian.com/User-Guide_Allwinner_overlays/ ... we might make an exception here, but now is as is.

igorpecovnik, 2017-10-29:
https://github.com/armbian/config/commit/550fcdfd66b839dee75db1f95670fe90bde71b3f @zador-blood-stained Which implementation path shall we go in such cases? usbhost0 and usbhost2 seems like a good to enable by default on Orange Pi Zero2 board for example. Shall we define a new variable in the board configuration for this since its board related? DEFAULT_OVERLAY="" ?

zador-blood-stained, 2017-10-29:
> Which implementation path shall we go in such cases?

IMO the best implementation is letting users read the documentation (mainly the SoC specific README shipped with overlays) and letting them edit /boot/armbianEnv.txt manually.
Many overlays have required parameters so adding those overlays without handling such parameters is kind of useless since users would have to read the documentation and edit armbianEnv.txt anyway. 

Ideally we could have some machine-readable format for overlay description (like json) and a dialog-driven menu for adding overlays, changing their parameters and displaying documentation, but this would take a lot of effort to make.

> usbhost0 and usbhost2 seems like a good to be enabled by default on Orange Pi Zero2 board for example.

Still it doesn't have any USB sockets besides microUSB OTG, so I'm in favor of leaving them as is (disabled) but work on better download pages and documentation to let users know why additional USB ports don't work by default.

igorpecovnik, 2017-10-29:
It seems that people are attaching USB to the header. First four pins are USB and in stock kernel and all our previous, this was enabled ... by accident or not. That's why I am thinking this is good. Also for g_serial.

zador-blood-stained, 2017-10-29:
We could think about default overlays later (no stable mainline images for H3 and especially H5 yet) and I agree with enabling g_serial on more boards by default. In any case it's not related to armbian-config.

igorpecovnik, 2017-10-29:
ok. The issue OP was resolved.

closed 2017-10-29 by igorpecovnik