armbian/build #807: gcc-arm-linux-gnueabihf > 7.0 not fullfilled on Xenial

rozie, 2017-10-21:
Hi.
Tried to build a full image for Orange Pi Zero (Jessie) in VM created for Ubuntu Xenial

Got message:
Could not find required toolchain [arm-linux-gnueabihf-gcc > 7.0]
I have newest available version from Xenial repo installed and it's 5.3.

igorpecovnik, 2017-10-21:
We don't use distribution compilers but external which should be downloaded automatically. Is your setup pure virtualized Ubuntu Xenial X64? It is strange since I restart my build environment from scratch about 1-2 weeks ago and missing dependencies were fixed on the way. We rebuild all kernels daily and all is normal. Well, I'll take a look ...

zador-blood-stained, 2017-10-21:
Last time toolchains were hosted on dl.armbian.com because releases.linaro.com was too slow from time to time. "Were hosted" is the problem because I don't see them on dl.armbian.com now.

igorpecovnik, 2017-10-21:
They are. I only mask them.

igorpecovnik, 2017-10-22:
I made another test from clean Xenial - Debian Jessie with 3.4 or NEXT compiles without a problem. I am recommending removing external toolchains to force re-download: rm -r cache/toolchains

closed 2017-10-22 by igorpecovnik

rozie, 2017-10-28:
Word of explaination:
I found the reason. When I installed Ubuntu, I didn't choose any role, so it was very base system. Didn't even had _curl_ or _wget_ installed. It may be worth to install required packages in the script or at least make notice in readme which role should be chosen.

Another thing: had to install _aptly_ as not trusted package.

igorpecovnik, 2017-10-28:
>Another thing: had to install aptly as not trusted package.

wget is needed to get a key ... ok. we will harden this thing a little bit more.

rozie, 2017-10-28:
Another missing packages I remember: xz-utils, pkg-config, pkgconf.

https://zakr.es/packages_armbian.txt - here's list of installed packages on VM where I sucessfully built Jessie image in case you find it useful.

kanghan90, 2018-10-11:
after clean install of Ubuntu. I had exactly same issue checked cash 7.0 tool-chain wasn't there now running compile.sh again lets see if it can download this time